repetier/RepetierFirmware #871: PWM does not work

Uka4me, 2019-05-10:
Hello!
I can not get pwm to work correctly.
I have mega and ramps 1.4. I want to control the output of the signal through the command M42 P4 S *, but I don’t keep silence. It always gives 5V or 0V if you specify M42 P4 S0.
I also tried other outputs: D11, D6, D5, D4, D42, D44.
I tried mega on a new one, because it was the same. I also tried different firmware versions: 0.92, 1.0.3, 1.0.4dev.
At the end of last year, as soon as I got a new sandwich, I flashed it and checked the work of the M42 team, now I’ve got to the board again, but I can’t get the team to work properly anymore ...
Please help...
[Configuration.txt](https://github.com/repetier/Repetier-Firmware/files/3167309/Configuration.txt)

Uka4me, 2019-05-10:
Sorry, Google Translate ..

Uka4me, 2019-05-10:
Any number >0 leads to the conclusion 5v.

RAyWB, 2019-05-10:
PWM was taken away from m42 due to conflicts with hardware pwmsettings.
it´s just commented out in commands.cpp

Uka4me, 2019-05-10:
Is there a workaround?

RAyWB, 2019-05-10:
not shure if it makes trouble in each setup , just comment out line 1931 in commands.cpp and recompile
should then work again. in best case your setup works without issue

Uka4me, 2019-05-10:
I found this place and corrected it, but it still did not help (( it turns on at 255, and turns off at <= 254.

Uka4me, 2019-05-10:
Well, since this place causes problems, is it possible to use another control method? For example, I have a laser and need TTL control from 0 to 5 volts. If the M42 command does not work as PWM, then what pin can I specify for M3? Or does the laser support end?

RAyWB, 2019-05-10:
m42 is usually not used for laser
and laser is supported on/off only from standard firmware 
, see forum about that stuff there are lots of threads

Uka4me, 2019-05-10:
In general, thank you for your help. You helped me a lot.
But in the end I made changes to the files: Commands.cpp and Drivers.cpp (for the M3 team). But it never worked for me ((
I found the firmware version 1.0.2, corrected the configuration and made the same changes in Drivers.cpp. In the end, everything worked fine for me. A search on the forum brought me to pin 44, as a priority in use for the laser.
Maybe later, when I have time, I will study the list of changes and try to make the firmware version 1.0.4dev work as well, and she does not want to respond to analogWrite. I also tried cleaning the pre-eeprom.
Maybe I do not perform what action previously? because there is a feeling that he simply does not see the changes and works as before on all pins ...
Although on version 1.0.2 my pins refused to work: D11 and D4. They wrote about d11 that he wasn’t working as pwm and everyone was talking about d4, but he didn’t work for me either. Tested only D5, D6, D44.

Uka4me, 2019-05-10:
By the way, another moment. When testing d5 pin with S255 value, it gives out 5 volts, and D44 with S255 gives only 3 volts ...
Power only from the company, tomorrow I will try to deliver more.
Maybe you have thoughts on this?

repetier, 2019-05-15:
3V on a 5V board is not correct. All pins are 5V. Have you tested if reducing pwm sets value down? Otherwise an other function might control the button as well, e.g. fan, heater, ... so it influences your value.