repetier/RepetierFirmware #873: V2 firmware

qiwalker, 2019-05-12:
Hi 

How to configure and setting  a and b axis in repetier v2 
Im totally new in repetier

Thanks
Rizki

repetier, 2019-05-13:
Mainly follow https://docfirmwarev2.repetier.com/config/motors

Set NUM_AXES 6 and define A and B motor, that is all. Maybe define end stops. Depending on why you need 5 motion axes you might also need to change the printertype. By default A,B,C are just extra motors not used for motion computations so they only get a basic handling to follow a linear axis.

qiwalker, 2019-05-13:
awesome!
can i use due board only with repetier ? 
i read repetier use EEPROM memory and due dont have it :(

thanks
Rizki

repetier, 2019-05-13:
Due board is the only one working correctly. If you use it e.g. with RADDS board which has an eeprom chip to compensate the problem, that would be no problem. Wir due alone no eeprom possible of course.

qiwalker, 2019-05-13:
#define NUM_AXES 4                   // X,Y,Z and E for extruder A,B,C would be 5,6,7
#define STEPPER_FREQUENCY 250000     // Maximum stepper frequency.
#define PREPARE_FREQUENCY 2000       // Update frequency for new blocks. Must be higher then PREPARE_FREQUENCY.
#define BLOCK_FREQUENCY 1000         // Number of blocks with constant stepper rate per second.
#define VELOCITY_PROFILE 5           // 1 = linear, 3 = cubic, 5 = quintic velocity shape
#define Z_SPEED 10                   // Z positioning speed
#define XY_SPEED 100                 // XY positioning speed for normal operations

should i add #define A_SPEED ? for "a" axis speed

and how to setting like step/mm or step/rev fro stepper motor

Thanks
Rizki

repetier, 2019-05-13:
So far good.
You need to copy definition for example x motor just with A as motor name in configuration_io.h. Same for defines in configuration.h. It you don't it will not compile and you see which defined are missing.

skwhite16, 2019-09-24:
Hi 

I have used v1 in the past and I am trying to get started on V2. I am using due / RADDS. I followed the documentation but now I am stuck. The code will upload successfully with #define MOTHERBOARD MOTHERBOARD_FELIX but the process is terminated if I change it to #define MOTHERBOARD MOTHERBOARD_RADDS. The problems are as follows:
[image]
Any advice?

repetier, 2019-09-25:
> image

Hi,

in HAL.cpp for due replace

staticinlinevoid noTone(uint8_t pin) {

with

staticinlinevoid noTone() {

and it should compile.

-- 
Mit freundlichen Grüßen/Yours

Roland Littwin

----------------------------------------------
Hot-World GmbH & Co. KG
Knickelsdorf 42
47877 Willich
Deutschland

Tel.: 02154-606518
Fax: 02154-606519

Geschäftsführer: Marcus Littwin
Handelsregister: Amtsgericht Krefeld HRA 5325
Umsatzsteuer-Identifikationsnummer: DE213748632
Persönlich haftender Gesellschafter: Hot-World Medien GmbH mit Sitz in Willich
Handelsregister: Amtsgericht Krefeld HRB 10800

closed 2019-09-25 by repetier

skwhite16, 2019-09-27:
I searched for the line that you are referring to and there is no such line in HAL.cpp... but I was able to get it to compile by changing the line in radds.h (line 172)

#define BEEPER_PIN 41

to 

#define BEEPER_PIN -1

But still, I haven't been able to get any motors to move yet...

repetier, 2019-09-28:
Maybe you have defined them to only move after homing?
```#define MOVE_X_WHEN_HOMED 0
#define MOVE_Y_WHEN_HOMED 0
#define MOVE_Z_WHEN_HOMED 0
#define MOVE_A_WHEN_HOMED 0```

Would cause it to only move after homing. Also check M119 to ensure no end stops are blocking. At least for x,y, z the demo config should work out of the box apart from these restrictions.