repetier/RepetierFirmware #878: Error freeze all extruders when installing k or L factor >0 (v1.03, v1.04dev)

Goodfeat, 2019-06-17:
Hello. I caught a new bug on firmware 1.03 and 1.04.

The error is connected with the K-factor and L-factor. 
If you set one of the factor > 0, an error may occur that causes all extruders to hang. The extruder will not respond to any commands. 
In this case, only the power off helps. Another forced reset helps several times in a row.

At first I thought that this drivers overheat. However, when the extruders hang, the signals to the driver outputs also do not pass (step, dir).

On firmware 1.04 it appears almost instantly. On firmware 1.03 this may appear after several switchings of the extruder (T1> T0> T1> T0 ...)

An interesting fact: when the extruders hang, the extruder counter continues to count the filament feed.

I suppose that the hangup is connected with the calculations of the K or L coefficients. Probably one of the variables is out of range.
For some reason, the firmware stops generating impulses to the driver. But it continues to count the amount of spent plastic
Perhaps in the firmware one of the flags can not perform a reset.

What could be the cause of the extruder freeze? What place in the code could theoretically cause extruders to hang when the coefficient is on?

(Delta printer, ruramps, Arduino DUE, driver LV8729 XYZ 640 step/mm, E0 / E1 2223 step/mm)

Goodfeat, 2019-06-17:
It seems to me that I found the cause of the freeze. But I have no idea what to do with it now.

void EXTRUDER_TIMER_VECTOR()
this function stops being called, so the extruder stops working.
I tried to put the function in commandLoop () to confirm my guesses.
The filing function is working. But with some problems and little freezes.
However, this is only a small part of the fault.

There was a question. Why does the interrupt freeze?
I lowered the STEP_DOUBLER_FREQUENCY speeds to 40,000
Increased the value of ALLOW_QUADSTEPPING, DOUBLE_STEP_DELAY, STEPPER_HIGH_DELAY to 2
This absolutely does not help in solving the problem.

repetier, 2019-06-17:
It should not be possible to interrupt that interrupt since it is in an interrupt protected block.

extruderChannel->TC_RC = Printer::maxExtruderSpeed;

sets the next timer value. If that would be missed it may take some time to get there again. Had this with main stepper and a missed interrupt meant there 4 minutes for next try:-(
Only other reason not to do anything is advance is disabled.

You should not use quadratic advance. In V2 I have removed it completely. It grows just much too fast and can then easily cause problems. Better only use linear advance in 1.0.4 which has some fixes.

Just some prior hints. Need some more time to be able to check it closer at all.

Goodfeat, 2019-06-17:
Thanks for answering

This is the point. L-factor freeze the same way. It also interrupt.
Can this be fixed?
it gives a lot of problems

Goodfeat, 2019-06-17:
It seems the problem with the timer is related to the settings of the extruder engine.
I began to suspect the setting of EXT0_STEPS_PER_MM 2223
Can a high value of STEPS_PER_MM cause the timer to skip?
I noticed that the void EXTRUDER_TIMER_VECTOR function is called more often when plastic is used. -at least 4-5 times more often than usual. 
I reduced the values to 500 mm. freezing seemed to dissappear.
I gradually began to increase the value.
It turned out the maximum limit of 800mm, at which the extruder does not freeze.
I use the LV8729 driver with microstep 128. At the same time, I have an extruder of 1: 3. 
Therefore, I need such a high value (2223)

Is this a firmware limitation?
Have I really achieved the maximum performance of the Arduino due?

Nibbels, 2019-06-17:
I experimented a lot with such behaviour.

But I cannot really talk here because I used a 8bit ATMega2560 - 16mhz. Plus I use some selftempered version of a very old repetier firmware which I upgraded as a hobby. So this is theoretically a whole other world.

When I set my extruders stepper to Microsteps=128 to gain about > 2000 steps/mm I gain a lot of problems communicating through serial. I get corruptions then. Those corruptions happend randomly having some chance like 1 per 10minutes.
So my experience is that you should never try to reach maximum interrupt frequency possible. What helped me alot was limiting the STEP_DOUBLER_FREQUENCY heavily. The solution for me was to always limit the interrupts frequency to a reasonable limit.

Using Advance you have many more interrupts because this uses another additional interrupt than the normal interrupt for movement. I suggest you take that into account when setting the STEP_DOUBLER_FREQUENCY. -> Set it lower than without advance.

To gain a better understanding of Double-/Quad-Stepping I flip the variable of STEP_DOUBLER_FREQUENCY to its inverse: minimum clocks between two single interrupts == output steps.
-> So on my processor I began to pack outgoing steps into one interrupt if the interrupt-speed falls below ~3000clocks distance inbetween single interrupts. That actually means that we reduce the microsteps by software to gain higher speeds.
(And I, personally, dont double step and quad step. I like to do single, double, tripple, quad ...)
In case I still get the numbers together my number of 3000 means something like 16.000.000/3000 = 5300 double step frequency.

Why limiting to minimal 3000 clocks = maximal 5300 doublestepfrequency?
To always have enougth processing time for the main loop and other interrupts. So I have a very fluent main loop which can process communication and queuing of moves fast. (Because of historical reasons my firmware is packed with expensive features a bit more than repetiers 1.04.)

Google tells me that at least some Due is running at 84mhz. So that is around 5x my speed. If I would scale the same way that would calculate to 28k STEP_DOUBLER_FREQUENCY.
That does not mean that is totally reasonable or right. Just as a hint.

There is one more thing to understand:
Lets say that we increase the speed from lets say 10mm/s to 200mm/s in small steps. Having a nicely set STEP_DOUBLER_FREQUENCY the result speed scales nearly linear. Because in case you nearly reach a too high interrupt frequency the step shifting kicks in and boosts out two+ steps per interrupt. -> saving a lot of calculation time.
If the STEP_DOUBLER_FREQUENCY is set too high there will be levels of speeds which produce a lot of additional workload. This is like driving a car at 50km/h using the first gear. In otherwords: "Your main loop and other interrupts cannot breath" driving some specific bandwidth of speeds.

-> My personal conclusions: 
1) Interrupt workload introduced problems might especially appear when you set specific ranges of speeds. You should have a nicely evaluated STEP_DOUBLER_FREQUENCY to avoid problematic speed bandwidths.
2) Always find the sweet spot for microsteps and dont necessarily use 0.9° Motors like I did :D
Very high steps/mm at an extruder is only good if you want to use a 0.15mm nozzle and 0.05 layer height.
3) The higher speeds you drive the more Advance will help your quality in edges. I think there is no need to have advance at 30mm/s. However if your material is very soft advance might still be some tweak.

Greetings

Goodfeat, 2019-06-17:
Thanks for the answer,
I relied heavily on DUE's computational abilities, and I really thought that Due's performance would be enough to calculate such large values.
I thought ~ 2000 was small enough for the printer to process well.
I had a 1:40 gearbox before the extruder 1: 3. An insane value EXT0_STEPS_PER_MM> 24,000 was committed. Nevertheless, the Arduino DUE processed this value as well. But before that I did not use the L-factor. Therefore, step = 2000 seemed to me a rather low value .
Now I understand that for some calculations like Linear advanced performance may not be enough.

I realized that I needed to experiment with the values ​​of STEP_DOUBLER_FREQUENCY
Thank you for sharing your experience)

repetier, 2019-06-18:
A hint for due speed - it is not 5x faster. It is also 32 bit so many operations are much faster then on 8 bit so I think it is in effect at least 10x faster. If you do not want double stepping up to 80KHz are ok. With double/quad stepping the interrupt time increases so it must be called not that often. It also depends on stepper drivers - if they require delays the speed quickly drops down and yes, there are drivers being slower then due can set signals!

Especially with advance your steps per mm should not be too high. The stepper interrupt is limited and it needs to catch up with advance move which can be impossible for high resolutions. Also depends on extruder speed but since you have problems with extruder switch it might be that moves are faster then stepper interrupt gets called?

Goodfeat, 2019-06-18:
Hello . Switching the extruder is fast enough. But I do not know where the switching of the extruder is regulated.
Acceleration and movement of stepper motors -3000 mm / s
But I noticed a glitch in the work of switching.
When I switch extruders, the first switch is slow, the subsequent switch is faster.
This happens if you use Z-Lift (I do not use it. During printing, it periodically gets buggy and does not return the extruder after lifting).
// ---
I tried various STEP_DOUBLER_FREQUENCY parameters, but it did not help me.
At the moment I have reduced EXT_STEPS_PER_MM to 570 by reducing the microstep to 32.
I came to the conclusion that 128 microstep for an extruder is not necessary.
The noise from the motor of the extruder remained almost at the same level

Now I am trying to achieve printer stability at 1.03. As soon as I set up the printer, I will switch to 1.04 again, I will catch bugs further.
would like to try repeater 2.0 but I can’t configure motors and enstopes. motors do not move. endoscopes are not working properly.
I can not understand the documentation. it seems it is not enough to adjust as it should be.

Thank you)