repetier/RepetierFirmware #884: [Feature Request] Marlin 2.0 DCE on ARM requires DTR be asserted, **always**

gruvin, 2019-07-28:
On Mac (and presumably Windows) the USB DCE implemented by Marlin `bugfix-2.0.x` requires that DTE DTR be always asserted, to enable communications. Without this, the Marlin DTE neither accepts or sends data, even though the port opened successfully.

RTS/CTS handshaking is not required or enabled, afaik.

These 32-bit boards do not implement any kind of USB serial bootloader. Thus, toggling DTR on then off is inappropriate. Rather, they seem to require standards compliant use of DTR, which of course is to assert DTR when the port is opened. I presume DSR is asserted in response, though in this case I see no reason to actually wait for DSR, since this is not a "real" serial port.

As it happens, this issue does not present itself with Pronterface, which seems to work just fine.

Obviously, the issue could be resolved with changes to Marlin. However I believe that would require messing with the lower level USB device code, outside of Marlin's normal scope. (Educated guess.)

repetier, 2019-07-29:
Is this testable with Arduino Due or which hardware is required to test that.

Have you tested Repetier-Server for Mac? There you can set DTR to be always high.

gruvin, 2019-07-30:
Sorry I don’t have. ‘DUE. 

However, it also occurs to me that even the Arduino 8bit boards do not require DTR to be pulsed to operate correctly. There is a capacitor in the circuit that causes only a short reset pulse, even when DTR remains high, which usually just happens as a matter of course, when the port opens. You could test that at your end, I presume?

The obvious concern would be breaking existing compatibility but I believe you’d be fine there, unless you know something I don’t about some odd board out there.

I wasn’t aware there was a server app. I’ll check it out when I get home.

Thanks.

gruvin, 2019-07-31:
OK, home and just activated a trial install of Repetier Server on my Mac. Firstly, what a beautiful bit of software! Looks great and is very smooth in function. Well done!

Connecting and transferring commands successfully to the SKR v1.3 board running Marl bugfix-2.0.x ...

- DTR Low to High works
- DTR High to Low does not work
- DTR Low does not work
- DTR High works

This is exactly as I would expect.

I suppose for maximum compatibility, Repetier Host should have the same DTR options, no? However, is Repetier Host still in active development or are you phasing it out in favour of Repetier Server? Personally, having now experienced it, I would be more than happy to use a local only (no remote control) version of Repetier Server, rather than Repetier Host. It's just that good, imo.

repetier, 2019-07-31:
Ok, so DTR high is the solution. I know other boards requiring RTS high, DTR low. Arduino 8 bit boards do not care as any change just does a reset. So it is really becoming complicated to serve all with one setting.

Regarding server - no need for external servers. If you want it local only you can tell firewall to not allow exit traffic on 3344. Server is in deed the planned replacement for mac host. It has the same code then windows and linux version, so it does not matter that are not so much mac users. Only thing it really misses is an integrated slicer which is also planned.

Mac host will not get much extension but small changes like dtr problem will still be done some time.

gruvin, 2019-07-31:
Great stuff. Thanks! 😄

closed 2021-09-09 by gruvin