repetier/RepetierFirmware #889: Adding TMC2208 (UART) Driver Support

Goodfeat, 2019-11-19:
Hello. TMC2208 drivers are controlled by the UART interface.
Is it possible to add support for TMC2208 drivers in firmware (1.04)?

repetier, 2019-11-19:
No and V1 branch will not get them.
However V2 branch already supports them along with 2130 and 5160 model.

weed2all, 2020-01-16:
Hi...I'm new to repetier...I have a due and radds..and some tmc2208!how can i use them in uart mode with repetier v2?

repetier, 2020-01-16:
Yes you can, also it has one problem. You need one hardware serial per driver. Have just added an update that allows you to combine multiple drivers by setting
#define IGNORE_TMC2208_FEEDBACK

This will disable reading stepper state so if all are programmed blindly with same parameters it should work (have only one driver, so can't test it with multiple drivers, but TMC2208 datasheet suggests this solution for multiple drivers).
So with this constraint it is possible. There is one serial on the outermost extension port.
If you need feedback for overtemperature you bette ruse TMC2130 where the problem does not exist.

weed2all, 2020-01-16:
Then j think the best will be to go with tmc5160!as it has more power than tmc2130...and with spi instead uart like tmc2208

repetier, 2020-01-16:
If you need the power yes. tmc5160 works as well.