repetier/RepetierFirmware #905: firmware for arduino mega 2560 does not compile ( v 2.0)

Goodfeat, 2020-02-20:
Hello, I decided to try firmware 2.0 again. but I still can't even compile it.

I did according to the instructions

[image]

changed the motherboard to ramps
[image]

added a line to the constant.x file
[image]

It seems that this should be enough for the firmware to compile so that I test it.
but errors are streaming in the terminal and the firmware does not compile for the desired dress.
[image]

[image]

please tell me why the firmware does not compile and how can I fix it?

repetier, 2020-02-20:
The 8 bit HAL required for HAL is not finished. It is already there from old V1 but not adopted to new requirements. Currently only due based boards, RUMBA32 and Adafruid Grand Central M4 +Board are supported and will compile. AVR will take a bit more time.

Goodfeat, 2020-02-20:
Does this mean it's useless to install firmware on arduino mega now?
I'm sorry, I was set up to test the new firmware.

repetier, 2020-02-20:
Yes, it is not compilable for AVR at the moment. We just have started adding the files but not finished developing them. That is why you see them already but can not compile.

Goodfeat, 2020-02-20:
OK

I hope it will work soon
I will wait for the firmware update