repetier/RepetierFirmware #907: dev2 - Endstop and probe behaviour not well documented

kyrreaa, 2020-02-29:
Since there is no config tool for dev2 yet it is very hard to configure.
Behaviour seems strange for end stops when referencing one for use as zprobe.

Clarification would be good for how end stops now work in dev2 with respect to defining, combining, polarity and usage for probing.

repetier, 2020-03-01:
Have you read https://docfirmwarev2.repetier.com/config/bed_calibration and https://docfirmwarev2.repetier.com/config/endstops ?
EndStop is just a base class module signaling a state and remembering it. It can be assigned directly to a global axis or used as motor endstop or as signal for z probe. If variable ZProbe is pointing to an endstop we assume zprobe is available and can be used for z min homing, no extra work required here.

Polarity is always same for all functions in firmware. High signal is triggered. Use inverting signal of input or pullup variant to ensur ethis condition wherever needed.

If you see parts in documentation that are not clear enough please let me know. I'm not a writer but at least I try to describe each module as soon as I write it to keep it always up to date.

kyrreaa, 2020-03-01:
I found the new config examples in the latest branch. My branch was stale sorry.

closed 2020-03-01 by kyrreaa