repetier/RepetierFirmware #908: dev2 - Max extruder feedrate appears to be ignored

kyrreaa, 2020-02-29:
With max tool 1 feedrate set to 100mm/s it still defaults to 10mm/s.
Using command M203 E6000 unlocks the speed until reset.
Setting value in eeprom appears to work and reading it back shows new value, but it is not being used.
M360 reports what is in the eeprom and it does not change when using M203.

repetier, 2020-03-01:
Have you updated your firmware? The problem was fixed at 27.1 commit as far as I know.

kyrreaa, 2020-03-01:
Thanks, I think I fubared the branch I was working on. somehow I have old dev2. Will update and see.

kyrreaa, 2020-03-01:
Ahh I figured out what happened. I went back to my older branch as the new one does not work with userpins.h due to a redefine of MOTHERBOARD_USER_DEFINED to MOTHERBOARD_USER_DEFINED_DUE and MOTHERBOARD_USER_DEFINED_AVR while still needing MOTHERBOARD_USER_DEFINED in due/pins.h to include userpins.h hehe.

closed 2020-03-01 by kyrreaa

repetier, 2020-03-01:
Thanks. Found remining MOTHERBOARD_USER_DEFINED and fixed it for next update.