repetier/RepetierFirmware #914: dev2 - core-xyz - Homing causing extruder motion?

kyrreaa, 2020-03-12:
Sometimes when I home it extrudes filament at rapid speed.
Is this connected to the de-select/reselect of tool or is there something else here?
Also, homing speed motions are sometimes wrong.

kyrreaa, 2020-03-12:
If extruder is cold the feed does not happen. I did remove any feed settings for retract/advance and any such firmware settings.

AbsoluteCatalyst, 2020-03-13:
I've sometimes had the extruder issue after long prints. I _think_ it might be related to switching between relative positioning and absolute positioning, then homing. It appears the extruder attempts to move back to what it was at the beginning of the print. I usually run all my prints with relative extrusion. 

I myself haven't experienced any other issues with homing speeds though.

kyrreaa, 2020-03-13:
The homing speed is usually correct but the back-move is often wrong after a print job. Resetting the printer (using reset button or emergency command) restores it for a while and back-move happens at full motion speed instead of a crawl.

AbsoluteCatalyst, 2020-03-14:
Do you happen to get a warning about an illegal position prevented when the homing speed bug happens by any chance? 
Also by back-move, do you mean ENDSTOP_X_BACK_MOVE etc? 

I'm not sure how I never noticed but I have the homing speed bug as well. I can reproduce it by homing my Z axis, then immediately trying to move my Z axis.

The following might be a bit of babble if it ends up not helping your issue at all; though it's what i've found so far: 

Reading the console after homing my Z, I notice it tries to move my X and Y axis back to the previous position (not homing) but fails as a illegal position. The previous position is my min X and min Y position (which is also the position after homing these two, -28 and -10 respectively.). The reason it fails is because the position it actually tries to move is eg. "-28.00**316**", which is slightly beyond the minimums due to floating point inaccuracies. Under PrinterTypeCartesian.cpp and within the function positionAllowed, subtracting and adding 0.01~ to the mins and max respectively, fixes that issue. 

In the middle of writing this I realized I don't use ENDSTOP_X_BACK_ON_HOME. Setting this to 0.5 causes... even more weirdness for me? It also doesn't appear to update the position to eg. -27.5, -9.5 after homing, so it'll still have a failure moving to the previous position.

If positionAllowed fails it'll prevent a move but the current position will still be set to whatever the destination was meant to be. On the next move queued it'll still attempt to move to that position as well but at weird speeds. 

I _think_ it'll sometimes use the speed set by ENDSTOP_X_RETEST_REDUCTION_FACTOR? 
But most of the time it appears to use the new speed given. 

It looks like this is why repetier unhomes the axes whenever there's a out of bounds position given as well as the warning that the position shouldn't be trusted anymore.

Unfortunately turning on MOVE_X_WHEN_HOMED and co. makes our axes unable to move since occasionally the print bounds issue appears right after homing.

kyrreaa, 2020-03-16:
I synched with dev2 now and am only using code from there pluss config adjustments.
BABYSTEPS_PER_BLOCK was wrong in config and has been corrected to {1,1,1,1,1} as I have 5 axis total. x,y,z + e + a

kyrreaa, 2020-03-16:
I have discovered that the reported E position from M114 before homing is the amount of motion extruder receives when homing.
Somehow it is reset to E0 in homing stage and then updates to restore targeted value causing motion.

repetier, 2020-03-17:
Ok found the position. At end of homing I go to old position for unhomed axes but extruder selection had reset E position to 0 so it had redone that move. In my tests the problem had vanished. Test script:
```
M302 S1
G1 E50
G28
```

Haven't seen the different homing speeds so far but will keep an eye on it. It is not related to the E bug which was just a extra move at the end.

kyrreaa, 2020-03-17:
Speed appears consistent after I corrected the babysteps array. So far anyway.

repetier, 2020-03-18:
Great. So I close the issue for being fixed.

closed 2020-03-18 by repetier