repetier/RepetierFirmware #916: one axis is slower than the others in delta printer

mj996, 2020-03-16:
Hello everybody,I have a problem with my delta printer and I would be very grateful if you help me 
 to solve it.
I'm using Arduino due + RADDS with Repetier firmware.
The problem started when the Z stepper motor only moved in one direction.I've tried loads of methods non of them helped me to solve that.
I checked all of the endstops.They worked fine.
I checked the stepper drivers ,all of them had the same voltage 0.7 v .
and finally I found out that the problem was with the dir pin on the board
So I decided to use E1 socket for Z axis stepper motor.I change it in my firmware and i plugged the stepper cable and the driver into E1.when I swapped the Z axis with E1,the Z carriage moved correctly up and down but the speed of that carriage is lower than the others now.so the things that I did :

- If i command the printer after homing to lower the Z for 30 mm  what i observe is 15mm distance from Z carriage to Z endstop but for the others it's 30 mm(correct)..
I thought that maybe the steps per mm is wrong but it was right.
200 stp/rev , 1/16 micro stepping , 2mm belt pitch , 20 pulley tooth
I set steps per mm on 80 for X,Y and Z.

-  I  checked the deep switches under the RADDS board and All of the were set on 1/16 (MS1=H , MS2=H , MS3=L).

- I swapped the driver of E1 with Y but the problem was still on z axis.

- When I put the Z axis stepper cable and it's driver back in Z socket there was no sign of speed problem(it's speed was like the other axes)but I can't use it because of the defect dir pin .

- I adjusted the belt tension and the X and Y axes worked fine and accurate but for Z axis we've got problem yet.

repetier, 2020-03-16:
Apparently the driver is using a different microstep setting. Check the MS1, MS2, MS3 pins on the socket if they mirror the switch settings. Maybe one of the switches is not 100% in position and still gives wrong signal. Just test voltage agains GND and compare it with Z socket. 5V over usb is enough for the test.

mj996, 2020-03-18:
Thanks.You made my day.There's another thing now.When I home the printer and command to lower the Z for 10 mm The carriages move 80mm downward and this will  just happen for the first command after homing.
should I open another issue for it?

mj996, 2020-03-18:
I noticed that when I press home button in the Host my last Max z height number will appears but when the carriages hit the endstops my new Max Z height will appear in the Host.So when i command to lower the Z for 10mm it will lower the Z 10 mm from my last Max Z height number.I commanded M502 , M500 but nothing changed.

mj996, 2020-03-18:
> Apparently the driver is using a different microstep setting. Check the MS1, MS2, MS3 pins on the socket if they mirror the switch settings. Maybe one of the switches is not 100% in position and [...]

Thanks.You made my day.There's another thing now.When I home the printer and command to lower the Z for 10 mm The carriages move 80mm downward and this will just happen for the first command after homing.
should I open another issue for it?
I noticed that when I press home button in the Host my last Max z height number will appear
but when the carriages hit the endstops my new Max Z height will appear in the Host.So when i command to lower the Z for 10mm it will lower the Z 10 mm from my last Max Z height number.I commanded M502 , M500 but nothing changed.

repetier, 2020-03-18:
That is no bug I think. Check in host what you have set as z position for homing. My guess is that it is 70 less then what you really have. So it sends lower Z coordinates to printer and hence the move is longer then expected since firmware knows where it was. Here M502/M500 does not help since it is the host having wrong coordinates. It does not know correct homing position and does not read returned coordinates in all cases.

mj996, 2020-03-25:
Sorry for the delay.I checked that and you were right the Z home position was different in the host.Now I don't have that Homing problem but still when I press home button the Host (manual control part) shows number of Z 1mm less than what i set it to be.
I checked the printer setting in the host the Z home position was ok.There's another thing in those settings called printable height. I set that 1 mm less than Z home position.Can this cause the problem?
So when I command g1 z0 the distance between the nozzle and the bed is more than what I expected and if I home the printer again and command g1 z0 one more time the distance between the nozzle and the bed is different from last time.

Regards

mj996, 2020-03-25:
> That is no bug I think. Check in host what you have set as z position for homing. My guess is that it is 70 less then what you really have. So it sends lower Z coordinates to printer and hence the [...]

Sorry for the delay.I checked that and you were right the Z home position was different in the host.Now I don't have that Homing problem but still when I press home button the Host (manual control part) shows number of Z 1mm less than what i set it to be.
I checked the printer setting in the host the Z home position was ok.There's another thing in those settings called printable height. I set that 1 mm less than Z home position.Can this cause the problem?
So when I command g1 z0 the distance between the nozzle and the bed is more than what I expected and if I home the printer again and command g1 z0 one more time the distance between the nozzle and the bed is different from last time.

Regards

repetier, 2020-03-25:
Do you mean Print Area Height - that is a limit for positions in z direction. So that can be limiting it. 
On the other side as soon as you send explizit Z coodinate like G1 Z0 host takes that position same as firmware and firmware will do it correctly even if host had different value before. Host is just showing what it thinks you are at but relevant for moves is what firmware says.

mj996, 2020-03-25:
> Do you mean Print Area Height - that is a limit for positions in z direction. So that can be limiting it.
> On the other side as soon as you send explizit Z coodinate like G1 Z0 host takes that [...]

Yes I mean the Print area height or Printable height in the host .I can't understand why I get different distances from the bed( G1 Z0) each time I home the printer !!
Another thing that seems unusual is the weird sound from Z stepper motor when I turn off the printer

repetier, 2020-03-25:
Yes G1 Z0 should always move to same height at same xy position. If you had autoleveling active there are some ways to get different results if you home to z max which I think you do, right? When homing z max you need to move back from z endstop at the end so you can correct z over xy without hitting endstop - otherwise z would not adjust leading to wrong z.
But sequence
G28
G1 Z0
should always lead to exactly same position and height. If not the end stop is triggering at different positions or z moves loose steps (which you should be able to hear).

When you power off printer there is no power to make sound, so not sure what I should think about the weird sound problem. Also it would not start a move so why should it make any sounds.

mj996, 2020-03-25:
> Yes G1 Z0 should always move to same height at same xy position. If you had autoleveling active there are some ways to get different results if you home to z max which I think you do, right? When [...]

Yes I home to Z max.I set the Endstop distance after homing 5mm for all x,y,z axis in the firmware.
I can hear some knock sounds from y stepper motor which I can sense it's vibration on the tower.So if we consider this as loosing steps here are things that I checked :

- same voltages on the stepper drivers(x,y,z).
- same belt tensions
and here are the current speed settings for all x,y,z :
- Max. travel speed 300 mm/s
- Homing speed 100 mm/s
- Travel acceleration 3000 mm/s^2
- Print acceleration 1000 mm/s^2

by the weird sound from steppers when powering it off I mean right at the moment I press power button on my printer I hear that (fisss) sound and  I also can see some little movements on the belt of Z axes at that moment.

repetier, 2020-03-25:
Loosing steps is normally a high pitched sound. Knocking sound is more if you go back a step due to being held by some force. Can also be overheating driver then you hear it going off and on.

Dispowering sounds like no error. When motors loose power the can not hold microstep position and that can be heard sometimes.

mj996, 2020-03-25:
> Loosing steps is normally a high pitched sound. Knocking sound is more if you go back a step due to being held by some force. Can also be overheating driver then you hear it going off and on. [...]

The sound is like a car crossing some rumble stripes on the road . There's fan for cooling the stepper drivers so I don't think that's an overheating issue and I can't see any obvious backward movements in pulleys and carriages.

mj996, 2020-03-26:
> Loosing steps is normally a high pitched sound. Knocking sound is more if you go back a step due to being held by some force. Can also be overheating driver then you hear it going off and on. [...]

what issues can cause the carriages to go back a step?If they are held by some force ,which force can cause this?belts?
Are there any items to check to find the problem?

repetier, 2020-03-26:
Normally only if you move behind physical possible position. Then you hit something. E.g. endstop reachend and move toward endstop without endstop check. But that is so loud you know that it is that.

mj996, 2020-03-26:
> Normally only if you move behind physical possible position. Then you hit something. E.g. endstop reachend and move toward endstop without endstop check. But that is so loud you know that it is that.

ok I'm sure that the carriages don't hit anything (including endstop) .Can this happen because of the speed or acceleration?
current rtravel speed  300 mm
I lowered the speed in firmware but the actual speed of printer didn't change ! (I changed EEPROM and Host setting to match the firmware too)

repetier, 2020-03-26:
It can also be jerk too high - in that case you loose steps and with some luck it starts moving or it keeps staying. That is at start/end. Max speed is an issue mid move if you start loosing steps there.

mj996, 2020-03-26:
So here is what I observed.I lower the speed by the command G1 Z0 F3000 and I didn't hear shock or knock sounds anymore!

repetier, 2020-03-26:
50mm/s is not really fast for a delta. At which point do you hear the sound exactly? During full move/start/end? And why only Y axis. 
G28
G1 Z0 F6000

moves all axis equally so they should also sound all the same.

mj996, 2020-03-26:
> 50mm/s is not really fast for a delta. At which point do you hear the sound exactly? During full move/start/end? And why only Y axis.
> G28
> G1 Z0 F6000
> 
> moves all axis equally so they should also sound all the same.

I can't say this exactly but I tried
G28
G1 Z0 F6000
for 5 times and most of the times I hear the sound when the carriages are close to their endpoint (z=0) and when they are going up for homing I can hear another one at the middle of the way.
And it's just for Y carriage.I don't know why !

Thanks for all these helps.I'm so glad for your support.

mj996, 2020-03-26:
Here is my log in the host : 

15:20:51.066 : Info:Watchdog Reset
15:20:51.066 : Free RAM:80316
15:20:51.066 : Autoretract:0
15:20:51.066 : X:0.00 Y:0.00 Z:0.000 E:0.0000
15:20:51.066 : SelectExtruder:0
15:20:51.070 : Error:Format error
15:20:54.870 : Warning: Seems like we missed a ok, got a wait - continue sending.
15:20:54.874 : Error:expected line 1 got 11
15:20:54.875 : Resend:1
15:20:54.892 : FIRMWARE_NAME:Repetier_0.92.9 FIRMWARE_URL:https://github.com/repetier/Repetier-Firmware/ PROTOCOL_VERSION:1.0 MACHINE_TYPE:Delta EXTRUDER_COUNT:1 REPETIER_PROTOCOL:3
15:20:54.895 : Printed filament:0.59m Printing time:0 days 0 hours 12 min
15:20:54.895 : PrinterMode:FFF
15:20:54.899 : X:0.00 Y:0.00 Z:0.000 E:0.0000
15:20:54.899 : DebugLevel:6
15:20:54.900 : SelectExtruder:0
15:20:54.912 : Error:Wrong checksum
15:20:54.913 : Resend:11
15:20:54.918 : skip 12
15:20:54.918 : skip 13
15:20:54.918 : skip 14
15:20:54.918 : skip 15
15:20:54.925 : DebugLevel:6
15:20:54.928 : SelectExtruder:0
15:21:03.802 : SelectExtruder:0
15:21:03.802 : X:0.00 Y:0.00 Z:616.200 E:0.0000
15:21:37.077 : SelectExtruder:0
15:21:37.082 : X:0.00 Y:0.00 Z:616.200 E:0.0000
15:24:41.530 : Reduced visual quality for better framerates and to protect print quality.
15:24:41.530 : You can disable this in Config->Preferences->Basic Settings->Behaviour.

repetier, 2020-03-26:
Any reason you start with such an old firmware version? 1.0.4dev is what I would start with. Or even V2 firmware (only available on github). Could really need a tester for new V2 on delta and delta code is working already also I still need to calibrate my delta. But that is the printer type benefitting most of new V2 system and it has already RADDS support. Only thing is no config tool - you need to write 2 config files instead, but with sample configs this is not that difficult if you have some programming background.

mj996, 2020-03-26:
Oh sorry we accidentally switched to that version on our last edit of settings in firmware.Now we're runnig 1.0.4 dev but our problem is still there.
I changed the driver but the problem is still on Y axis and there's another problem now. When I home the printer all of the carriages start going up but Y carriage stops in the middle of the way !!

repetier, 2020-03-26:
Please try using the E2 socket for Y axis. Maybe there is a bad contact also it sound quite repeatable where things happen. And switch X and Y stepper driver so it is new socket with working driver.

Was the Y speed different? If it has double microsteps it might not reach the top. Otherwise only reason to stop is the y axis gets triggered. That can also be a problem for normal moves if you have ALWAYS_CHECK_ENDSTOPS enabled. Please make sure it is disabled. Maybe it is just crosstalk from endstop blocking a short segment on y axis.

mj996, 2020-03-26:
Ok I swapped Y with E2 and now there's no sign of knocking or any other sounds anymore.So what's wrong with my Y socket?

PS : I also disabled ALWAYS_CHECK_ENDSTOPS in firmware and for now that problem is gone.

repetier, 2020-03-26:
Great so it is hardware related. Either socket since swap helped or endstop related as you changed that as well. You can remove one of them if you want  to know which one was real cause.

mj996, 2020-03-27:
> Great so it is hardware related. Either socket since swap helped or endstop related as you changed that as well. You can remove one of them if you want to know which one was real cause.

I tested the always check endstops option and the problem was not that. It was with the Y socket.So does it mean that Y socket is defected? Can I use the Y socket anymore?
And there is one more problem,When I command the printer to go to a specified coordinates (G1 X-140 Y-84 Z25), it will go there but after that the printer doesn't do any more commands ,no matter It's home command or other specified coordinates.
and **Sometimes** when I disconnect the printer and connect it again and command G28,the printer first goes to the coordinates that I commanded last time(G1 X-140 Y-84 Z25) and after that it will go to home position!!

repetier, 2020-03-27:
> I tested the always check endstops option and the problem was not that. It was with the Y socket.So does it mean that Y socket is defected? Can I use the Y socket anymore?
I would avoid it if possible. If you need the socket use it for extrusion - there it is not as bad as for motion to miss a few steps. It's strange from the repeatablility, but can't say why it makes a difference. As you see the algorithms are working correct. So maybe it has a bad soldering with connection to another pin causing the problems. Would explain repeatablity.

Regarding commands - check the host log with ack/commands enabled. That sounds like communication stack is broken and after reconnect it sends the missend motion.

mj996, 2020-03-27:
> Regarding commands - check the host log with ack/commands enabled. That sounds like communication stack is broken and after reconnect it sends the missend motion.

The ack/commands was disabled so I enable it.Now we can say that the problem is almost gone but not completely.
G1 X-140 Y-84 Z25
printer moves
G1 X140 Y-84 Z25
printer moves
G1 X0 Y84 Z25
printer moves
Now we want to return to our start point So
G1 X-140 Y-84 Z25
printer doesn't move and here is what we see in the log

12:28:00.577 : Error:B hit floor
12:28:00.593 : Warning:Invalid delta coordinate - move ignored x:-8885 y:-5128 z:2000
12:28:00.593 : Warning:in queueDeltaMove to calculateDeltaSubSegments returns error.
12:28:00.593 : Warning:executeGCode / queueDeltaMove returns error
12:28:01.562 : Reduced visual quality for better framerates and to protect print quality.
12:28:01.562 : You can disable this in Config->Preferences->Basic Settings->Behaviour.
12:28:01.578 : wait
12:28:01.640 : T:-22.05 /0 B:5.85 /0 B@:0 @:0
12:28:02.578 : wait

And When We command G1 X0 Y0 z25 , first the printer wants to go to (G1 X-140 Y-84 Z25) coordinates that we commanded last time and after that it wants to run current command(G1 X0 Y0 z25).

repetier, 2020-03-27:
.. hit floor comes if one of the slider is closer then DELTA_FLOOR_SAFETY_MARGIN_MM to the floor. This sounds like you are going very close to the physical limits of the delta. Maybe even an area where you should not be. When one of the arms need to be nearly vertical that is a very bad position. Due to angle it needs to move much faster then the others and that can easily be too much. That is also where the sliders are closest to bottom.

mj996, 2020-03-27:
Ok I think we're in the hardest level of calibrating a delta which we should have the same distance from nozzle to bed in different spots of the bed.

So I started it by computing the appropriate Z height when we are at the center of the bed and Z=0 So the gap between the nozzle and the bed should be as the thickness of a paper (0.1mm with a bit dragging on the paper).I set it and till here everything was ok.

I understood that I have a concave surface because when I homed the printer and commanded (G1 X-140 Y-84 Z0) to move the nozzle toward X Tower ,the distance between the nozzle and the bed was about 60mm !! I think that's a huge error and something is wrong here because if I want to continue with these situation it means that I should lower my endstop too much.(~60 mm).
I even tried that and  I lowered the endstop but that was a mess because when I homed the printer,the effector was too much close to the edge.I don't know what's the problem?

( As far as I know the standard way of calibrating a delta is that we should have the correct Z height in the middle and then set the gap between the nozzle and the bed in different spots and after all of these when we come back to the center position we may face another distance from nozzle to bed ,from the one that we had first time and we finally  should fix this by changing the DELTA SMOOTH ROD OFFSET which is MAX DELTA RADIOUS in repetier firmware.Did I miss something?)

repetier, 2020-03-27:
Ok that explains a lot. The convex/concave motion when moving away from xy 0,0 means the ratio DELTA_DIAGONAL_ROD  and PRINTER_RADIUS is way off. DELTA_DIAGONAL_ROD is distance of the rotation points connecting slider with carrier. PRINTER_RADIUS is the horzontal distance of these 2 points when carrier is at 0,0 and hence much smaller then DELTA_DIAGONAL_ROD. While it is easy to measure diagonal it is hard to exactly measure PRINTER_RADIUS but it should be possible within a few mm so you do not get 60mm error.

Use escher delta calibration with 6 factors to calibrate the delta parameter further.

mj996, 2020-03-28:
> Ok that explains a lot. The convex/concave motion when moving away from xy 0,0 means the ratio DELTA_DIAGONAL_ROD and PRINTER_RADIUS is way off. DELTA_DIAGONAL_ROD is distance of the rotation points [...]

Thanks alot I'm working on it.
Since this morning I receive ''no start signal detected '' in repetier host log and I think that's a common issue among the users cause I found so many issues with this title.
and here are the things which I tried : 

- I changed the baud rate to other numbers but that didn't work.
- I set the PORT and TRANSFER PROTOCOL to Autoselect or autodetect andno change in result.
- I tried to upload firmware with new EEPROM set and that didn't work niether.(I get Done uploading message each time I try)
- I tried both native and programming micro usb ports on due and non of them changed anything I still get '' no start signal detected '' ( I tried to send M114 and M115 commands with both ports but I got '' 1 command waiting '' notification )

mj996, 2020-03-28:
The strange thing is that I connected to the printer so many times before this with current settings and I didn't change anything ,I don't know where did this problem come from !

mj996, 2020-03-29:
Ok here is an update : 

- I tried different cable and still nothing.
- I tought that maybe there's a problem with USB ports on laptop So I tried to connect with a PC. but that didn't work neither.( I tried different ports on PC and matched the COM and baudrate in the host with them).
- I disabled WATCHDOG in firmware too .
So I'm really confused and don't know what to do with it !

repetier, 2020-03-29:
Please try EEPROM_MODE 0 instead and see if it connects. When eeprom does not work it will block forever communication and firmware. Especially if you did not change anything eeprom is always a possible reason.

mj996, 2020-03-29:
Thanks a million . Now it works again. So does this mean that if I wanted to upload new settings later I should continue with EEPROM_MODE_0 ?

repetier, 2020-03-29:
EEPROM_MODE 0 means you have no eeprom to store data. If you set it to a new value not used last time like 5 and it again does not work then eeprom or connections to eeprom are defect. Since you already had problems with one axis it looks like the radds is badly soldered or you have some other problems. I mean eeprom worked a few days ago and now it seems defect.

If you always have a sd card inserted you can modify in pins.h in radds section
#define EEPROM_AVAILABLE EEPROM_SDCARD

then you use the sd card as eeprom replacement. Better then loosing all settings every reset.

mj996, 2020-03-29:
Ok I remeasured my parameters for calibrating and change them in firmware and put them in Escher calibration calculator and here they are : 
Steps/mm (for Repetier only): 80
Initial endstop corrections: 0
Initial diagonal rod length: 444
Initial delta radius: 202.76
Initial homed height: 618.84 
Initial tower angular position corrections:   X : 3      Y : 5     Z : 0
Printable bed radius: 170
Number of probe points: 7
Number of factors to calibrate:  6
Normalise endstop corrections?  yes   

and I also measured the Z height errors at different suggested spots : 

Point 0 | X:0 |   | Y:170 |   |   | Nozzle height error: 48
Point 1 | X:147.22 |   | Y:85 |   |   | Nozzle height error: 81.5
Point 2 | X:147.22 |   | Y:-85 |   |   | Nozzle height error: 52.5
Point 3 | X:0 |   | Y:-170 |   |   | Nozzle height error: 79
Point 4 | X:-147.22 |   | Y:-85 |   |   | Nozzle height error: 42.5
Point 5 | X:-147.22 |   | Y:85 |   |   | Nozzle height error: 72
Point 6 | X:0 |   | Y:0 |   |   | Nozzle height error: 0

and here are the results : 

New endstop corrections: | X:85 |   | Y:0 |   | Z:392 |  
New diagonal rod length: 444 
New delta radius:87.55 
New homed height:638.08  
New tower position   angle corrections: | X:9.12 |   | Y:8.41 |   | Z:0 |  

The thing that I don't undrstand is New endstop correction.Does this mean that I should  move my endstops that much physically?For example should I move my Z endstop 391mm Upward?!

PS : I measured my DELTA_DIAGONAL_ROD and PRINTER_RADIUS as precise as I could.

repetier, 2020-03-29:
As I said the delta radius was completely of and escher algortith sees it similar changin it from 202.76 to 87.55 so that is surely the main error.

Endstop corrections are parameter how many steps to go down after reaching end stop to center delta correctly. It is a eeprom parameter and in configuration.h I think it means these:
/** To allow software correction of misaligned endstops, you can set the correction in steps here. If you have EEPROM enabled
you can also change the values online and autoleveling will store the results here. */
#define DELTA_X_ENDSTOP_OFFSET_STEPS 0
#define DELTA_Y_ENDSTOP_OFFSET_STEPS 0
#define DELTA_Z_ENDSTOP_OFFSET_STEPS 0

Once you have adjusted you should see big improvement but also should redo calibration. Next iteration will change a lot less and it should converge to a stable set of values.

mj996, 2020-03-29:
Ok,after 6 attempts in escher calibration calculator I got error message the second attempt had good results as I got about 30mm errors at corners.but the fact is that the PRINTER RADIOUS in that attempt was too close to the one that I measured myself.(about 190 mm).I believe that my real PRINTER RADIOUS is somthing  between 200-202 mm.Further attempts had strange PRINTER RADIOUS (like 45.4mm , 12.5mm .etc !!).I sent some images that show an estimation of my measurements.Maybe I'm making a mistake of measurments that I can't understand it.

mj996, 2020-03-30:
I was thinking that maybe I put measurements in wrong places .There are two parameters in the firmware as ROD RADIUS and PRINTER RADIUS and there is one parameter in Repetier host called PRINTABLE RADIUS.I don't know which of them is DELTA RADIUS and what should I put in those two other parameters.

repetier, 2020-03-30:
ROD_RADIUS is the real horizontal distance as you measured it. If not set explicitly it gets computed like this:
```
#define ROD_RADIUS \
    (PRINTER_RADIUS - END_EFFECTOR_HORIZONTAL_OFFSET - CARRIAGE_HORIZONTAL_OFFSET)
#endif
```

/** Max. radius (mm) the printer should be able to reach. */
#define DELTA_MAX_RADIUS 200

That is the limits where firmware will refuse to move to, so this limits xy independent of what is physically reachable but should of course be less.

mj996, 2020-04-25:
> ROD_RADIUS is the real horizontal distance as you measured it. If not set explicitly it gets computed like this:
>
> ```
> #define ROD_RADIUS \
>     (PRINTER_RADIUS - END_EFFECTOR_HORIZONTAL_OFFSET - [...]

> ROD_RADIUS is the real horizontal distance as you measured it. If not set explicitly it gets computed like this:
>
> ```
> #define ROD_RADIUS \
>     (PRINTER_RADIUS - END_EFFECTOR_HORIZONTAL_OFFSET - [...]

Ok after several days I found the problem.DELTA FLOOR SAFETY MARGINE ! it was a high number that preventing the carriages to move downward to get to the correct coordinates.
While I was cheering for finding the problem suddenly another problem came out from no where.
I was checking the Vref of stepper drivers and I recognized  that the LED on due board went off and the Vref of the drivers decreased from 0.7v to 0.18v . I couldn't turn on my arduino due board anymore.I tested several things (i.e Regulator ) but non of them worked. 
So I thought that my due board is no more alive.I bought another due,connected it to the printer,checked the settings and I changed my microstepping from 1/16 to 1/32 with dip switches and my steps per mm for all steppers from 80 to 200 in the firmware.
I tested the G28 command but only my y axis started going up.I changed the X stepper wire with X and this time only X axis started going upward.next time I changed the X driver with Y and I saw that the Y axis is moving !!
I'm afraid that the same problem would happen to my new board too and I burn another one.
I couldn't figer out the problem So I would be very thankful if you help me to solve this.

repetier, 2020-04-25:
Did you ever move the motors by hand? That causes back power and can destroy electronics if you move too fast.
If you do not connect to usb do you have 5V and 3.3V on the pins where it should be? 
You can try and move Y stepper driver to X or Z and see if it works there. Or X and Z driver to Y, so not switching motors. Then you know the drivers are ok.
But for electronic problems I'm the wrong person - only limited experience here.

mj996, 2020-04-25:
> Did you ever move the motors by hand? That causes back power and can destroy electronics if you move too fast.
> If you do not connect to usb do you have 5V and 3.3V on the pins where it should be? [...]

Yes I moved carriages but only when there was no power in electronics and the cables were totally out of sockets.

So I placed the Y driver on X socket and now the X stepper works fine and the Y stepper won’t move So should we conclude this as a defective driver on Y axis?
Could it be a defect on Enable pin of driver?

I measured the voltage between EN and GND pins of all drivers and all of them were near to 5 volts.

repetier, 2020-04-26:
If driver works on Y all pins required on Y and driver are ok.
If same tested driver works on x axis also X pins are ok.
If untested X driver on working Y place does not work the driver is defect and must be replaced.
So says my logic.

mj996, 2020-04-27:
Ok the problem was with the driver or it's better to say with drivers !! I don't know how did I destroy 
3 drivers !! Maybe at the times the nozzel hit the bed and I powered off the printer after some seconds...
Anyway,something seems weird here.When I switch micro stepping from 1/16 to 1/32 something goes wrong because at 1/32 microstepping ,When I command to lower the Z for 10mm all the sliders actually move 13mm.As far as I know this can be from a wrong Resolution number in firmware or Loosen belts and I checked them and everything seemed ok to me.I set Resolution at 200 and set the tension on belts(I also checked dip switches on Radds and set them to 1/32 correctly) but I still have that problem when I'm in 1/32 microstepping and if I set it back to 1/16(and of course the resolution to 80) everything become ok.

RAyWB, 2020-04-27:
if your resolution is 80 steps/mm  at 1/16  you have to set it to 160 at 1/32 as you just double steps.
if you set to 200 you will get a move of 12.5mm instead of wanted 10mm

mj996, 2020-05-03:
> if your resolution is 80 steps/mm at 1/16 you have to set it to 160 at 1/32 as you just double steps.
> if you set to 200 you will get a move of 12.5mm instead of wanted 10mm

Oh You were right that was my fault.Thank you so much for that.
Ok replaced new drivers and set the microstepping to 1/32.
To continue the calibration I used escher calculator again .I tried the 7 factor mode several times but I couldn't get to zero error.So I tried 4 factor calibration and My best try was when I had -0.81 , 0.03 , 0.1  nozzle height error for Z,Y,X corners and -0.06 for the center.After this try I got worse errors each time.

So I decided to calibrate my delta manually step by step to find the problem :

1. I lowered the nozzle at center point and corrected the Z_MAX_LENGTH to have a nice gap between the nozzle and the bed at the center of the bed .(I had a nice and smooth drag  on the paper which I was crossing it under the nozzle)

2. I lowered the nozzle at the X,Y,Z corners (the 3  probe points which escher offered).These points were higher than the center so I have a concave surface and I should fix these offsets by these parameters in firmware : DELTA_X_ENDSTOP_OFFSET_STEPS , DELTA_Y_ENDSTOP_OFFSET_STEPS ,
DELTA_X_ENDSTOP_OFFSET_STEPS .At the beginning I thought that each of these parameters will only effect its axis but I was wrong.if you change Each of these parameters it will affect the other axes too.
anyway after several attempts I could get to a fair nozzle height at the corner points.

3.I lowered the nozzle at the center of the bed again and It was obvious that the nozzle height was different from the first time I corrected it.( I have a concave bed so the nozzle was to high from the bed this time).So I had to fix this by changing the  DELTA SMOOTH ROD OFFSET which in Repetier firmware this is the the ratio DELTA_DIAGONAL_ROD and PRINTER_RADIUS.

I'm sure about my DELTA_DIAGONAL_ROD cause I measured my Rods several times and they were all the same.
So I have to change the PRINTER_RADIUS to get to the good result.As far as I know if you want to  lower the nozzle  you should increase the PRINTER_RADIUS and vise versa.
So i increased the PRINTER_RADIUS for several times but that didn't do anything and I had the same nozzle height at the center.
No matter how much I increase the PRINTER_RADIUS it doesn't do any thing

So that's my problem and if I make this solved I have to lower the nozzle at the corner points again and I guess I will face less nozzle height and I should fix that by changing the  Z_MAX_LENGTH.

I would appreciate any help and if you see any mistakes in my calibration process please tell me.

PS : I have no EEPROM enabled because if I set that I will get the ''No start signal detected '' error at the host.

mj996, 2020-05-03:
> If driver works on Y all pins required on Y and driver are ok.
> If same tested driver works on x axis also X pins are ok.
> If untested X driver on working Y place does not work the driver is defect and must be replaced.
> So says my logic.

Ok replaced new drivers and set the microstepping to 1/32.
To continue the calibration I used escher calculator again .I tried the 7 factor mode several times but I couldn't get to zero error.So I tried 4 factor calibration and My best try was when I had -0.81 , 0.03 , 0.1  nozzle height error for Z,Y,X corners and -0.06 for the center.After this try I got worse errors each time.

So I decided to calibrate my delta manually step by step to find the problem :

1. I lowered the nozzle at center point and corrected the Z_MAX_LENGTH to have a nice gap between the nozzle and the bed at the center of the bed .(I had a nice and smooth drag  on the paper which I was crossing it under the nozzle)

2. I lowered the nozzle at the X,Y,Z corners (the 3  probe points which escher offered).These points were higher than the center so I have a concave surface and I should fix these offsets by these parameters in firmware : DELTA_X_ENDSTOP_OFFSET_STEPS , DELTA_Y_ENDSTOP_OFFSET_STEPS ,
DELTA_X_ENDSTOP_OFFSET_STEPS .At the beginning I thought that each of these parameters will only effect its axis but I was wrong.if you change Each of these parameters it will affect the other axes too.
anyway after several attempts I could get to a fair nozzle height at the corner points.

3. I lowered the nozzle at the center of the bed again and It was obvious that the nozzle height was different from the first time I corrected it.( I have a concave bed so the nozzle was to high from the bed this time).So I had to fix this by changing the  DELTA SMOOTH ROD OFFSET which in Repetier firmware this is the the ratio DELTA_DIAGONAL_ROD and PRINTER_RADIUS.

I'm sure about my DELTA_DIAGONAL_ROD cause I measured my Rods several times and they were all the same.
So I have to change the PRINTER_RADIUS to get to the good result.As far as I know if you want to  lower the nozzle  you should increase the PRINTER_RADIUS and vise versa.
So i increased the PRINTER_RADIUS for several times but that didn't do anything and I had the same nozzle height at the center.
No matter how much I increase the PRINTER_RADIUS it doesn't do any thing

So that's my problem and if I make this solved I have to lower the nozzle at the corner points again and I guess I will face less nozzle height and I should fix that by changing the  Z_MAX_LENGTH.

I would appreciate any help and if you see any mistakes in my calibration process please tell me.

PS : I have no EEPROM enabled because if I set that I will get the ''No start signal detected '' error at the host.

mj996, 2020-05-03:
Here is my config file ( movement and misc setting) : 

// ################# XYZ movements ###################

#define X_ENABLE_ON 0
#define Y_ENABLE_ON 0
#define Z_ENABLE_ON 0
#define DISABLE_X 0
#define DISABLE_Y 0
#define DISABLE_Z 0
#define DISABLE_E 0
#define INVERT_X_DIR 0
#define INVERT_X2_DIR 0
#define INVERT_Y_DIR 0
#define INVERT_Y2_DIR 0
#define INVERT_Z_DIR 0
#define INVERT_Z2_DIR 0
#define INVERT_Z3_DIR 0
#define INVERT_Z4_DIR 0
#define X_HOME_DIR -1
#define Y_HOME_DIR -1
#define Z_HOME_DIR 1
#define X_MAX_LENGTH 200
#define Y_MAX_LENGTH 200
#define Z_MAX_LENGTH 612
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define PARK_POSITION_X 0
#define PARK_POSITION_Y 0
#define PARK_POSITION_Z_RAISE 10

#define DISTORTION_CORRECTION 0
#define DISTORTION_CORRECTION_POINTS 5
#define DISTORTION_LIMIT_TO 2
#define DISTORTION_CORRECTION_R 100
#define DISTORTION_PERMANENT 1
#define DISTORTION_UPDATE_FREQUENCY 15
#define DISTORTION_START_DEGRADE 0.5
#define DISTORTION_END_HEIGHT 1
#define DISTORTION_EXTRAPOLATE_CORNERS 0
#define DISTORTION_XMIN 10
#define DISTORTION_YMIN 10
#define DISTORTION_XMAX 190
#define DISTORTION_YMAX 190

// ##########################################################################################
// ##                           Movement settings                                          ##
// ##########################################################################################

#define FEATURE_BABYSTEPPING 0
#define BABYSTEP_MULTIPLICATOR 1

#define DELTA_SEGMENTS_PER_SECOND_PRINT 180 // Move accurate setting for print moves
#define DELTA_SEGMENTS_PER_SECOND_MOVE 70 // Less accurate setting for other moves
#define EXACT_DELTA_MOVES 1

// Delta settings
#define DELTA_DIAGONAL_ROD 444 // mm
#define DELTA_ALPHA_A 213.55
#define DELTA_ALPHA_B 335.57
#define DELTA_ALPHA_C 90
#define DELTA_RADIUS_CORRECTION_A 0
#define DELTA_RADIUS_CORRECTION_B 0
#define DELTA_RADIUS_CORRECTION_C 0
#define DELTA_DIAGONAL_CORRECTION_A 0
#define DELTA_DIAGONAL_CORRECTION_B 0
#define DELTA_DIAGONAL_CORRECTION_C 0
#define END_EFFECTOR_HORIZONTAL_OFFSET 0
#define CARRIAGE_HORIZONTAL_OFFSET 0
#define DELTA_MAX_RADIUS 170
#define ROD_RADIUS 209
#define PRINTER_RADIUS 209
#define DELTA_HOME_ON_POWER 0
#define STEP_COUNTER
#define DELTA_X_ENDSTOP_OFFSET_STEPS 630
#define DELTA_Y_ENDSTOP_OFFSET_STEPS 1905
#define DELTA_Z_ENDSTOP_OFFSET_STEPS 0
#define DELTA_FLOOR_SAFETY_MARGIN_MM 60
//#define SOFTWARE_LEVELING

#define DELTASEGMENTS_PER_PRINTLINE 20
#define STEPPER_INACTIVE_TIME 360L
#define MAX_INACTIVE_TIME 0L
#define MAX_FEEDRATE_X 150
#define MAX_FEEDRATE_Y 150
#define MAX_FEEDRATE_Z 150
#define HOMING_FEEDRATE_X 100
#define HOMING_FEEDRATE_Y 100
#define HOMING_FEEDRATE_Z 100
#define HOMING_ORDER HOME_ORDER_ZXY
#define ZHOME_PRE_RAISE 0
#define ZHOME_PRE_RAISE_DISTANCE 10
#define RAISE_Z_ON_TOOLCHANGE 0
#define ZHOME_MIN_TEMPERATURE 0
#define ZHOME_HEAT_ALL 1
#define ZHOME_HEAT_HEIGHT 20
#define ZHOME_X_POS 999999
#define ZHOME_Y_POS 999999
#define ENABLE_BACKLASH_COMPENSATION 0
#define X_BACKLASH 0
#define Y_BACKLASH 0
#define Z_BACKLASH 0
#define RAMP_ACCELERATION 1
#define STEPPER_HIGH_DELAY 1
#define DIRECTION_DELAY 1
#define STEP_DOUBLER_FREQUENCY 80000
#define ALLOW_QUADSTEPPING 1
#define DOUBLE_STEP_DELAY 1 // time in microseconds
#define MAX_ACCELERATION_UNITS_PER_SQ_SECOND_X 1500
#define MAX_ACCELERATION_UNITS_PER_SQ_SECOND_Y 1500
#define MAX_ACCELERATION_UNITS_PER_SQ_SECOND_Z 1500
#define MAX_TRAVEL_ACCELERATION_UNITS_PER_SQ_SECOND_X 1500
#define MAX_TRAVEL_ACCELERATION_UNITS_PER_SQ_SECOND_Y 1500
#define MAX_TRAVEL_ACCELERATION_UNITS_PER_SQ_SECOND_Z 1500
#define INTERPOLATE_ACCELERATION_WITH_Z 0
#define ACCELERATION_FACTOR_TOP 100
#define MAX_JERK 5
#define MAX_ZJERK 0.3
#define PRINTLINE_CACHE_SIZE 16
#define MOVE_CACHE_LOW 10
#define LOW_TICKS_PER_MOVE 250000
#define EXTRUDER_SWITCH_XY_SPEED 100
#define DUAL_X_AXIS 0
#define DUAL_X_RESOLUTION 0
#define X2AXIS_STEPS_PER_MM 100
#define FEATURE_TWO_XSTEPPER 0
#define X2_STEP_PIN   ORIG_E1_STEP_PIN
#define X2_DIR_PIN    ORIG_E1_DIR_PIN
#define X2_ENABLE_PIN ORIG_E1_ENABLE_PIN
#define FEATURE_TWO_YSTEPPER 0
#define Y2_STEP_PIN   ORIG_E1_STEP_PIN
#define Y2_DIR_PIN    ORIG_E1_DIR_PIN
#define Y2_ENABLE_PIN ORIG_E1_ENABLE_PIN
#define FEATURE_TWO_ZSTEPPER 0
#define Z2_STEP_PIN   ORIG_E0_STEP_PIN
#define Z2_DIR_PIN    ORIG_E0_DIR_PIN
#define Z2_ENABLE_PIN ORIG_E0_ENABLE_PIN
#define FEATURE_THREE_ZSTEPPER 0
#define Z3_STEP_PIN   ORIG_E2_STEP_PIN
#define Z3_DIR_PIN    ORIG_E2_DIR_PIN
#define Z3_ENABLE_PIN ORIG_E2_ENABLE_PIN
#define FEATURE_FOUR_ZSTEPPER 0
#define Z4_STEP_PIN   ORIG_E3_STEP_PIN
#define Z4_DIR_PIN    ORIG_E3_DIR_PIN
#define Z4_ENABLE_PIN ORIG_E3_ENABLE_PIN
#define FEATURE_DITTO_PRINTING 0
#define USE_ADVANCE 0
#define ENABLE_QUADRATIC_ADVANCE 0

// ################# Misc. settings ##################

#define BAUDRATE 115200
#define ENABLE_POWER_ON_STARTUP 1
#define POWER_INVERTING 0
#define AUTOMATIC_POWERUP 0
#define KILL_METHOD 0
#define ACK_WITH_LINENUMBER 1
#define KEEP_ALIVE_INTERVAL 2000
#define WAITING_IDENTIFIER "wait"
#define ECHO_ON_EXECUTE 1
#define EEPROM_MODE 0
#undef PS_ON_PIN
#define PS_ON_PIN -1
#define STARTUP_GCODE "00"
#define JSON_OUTPUT 1

repetier, 2020-05-04:
Actually it is ROD_RADIUS that gets used for computation. In the original config it  get scomputed by subtracting CARRIAGE_HORIZONTAL_OFFSET and END_EFFECTOR_HORIZONTAL_OFFSET from PRINTER_RADIUS but not in config tool.

You should use the 6 point escher solution which keeps diagonal length constant. It is bad to adjust this as well as then object size will get wrong. Main problem is that each modification changes height at all positions. But some are symmetric and some have more effect in one region. So angles have an effect in neighbouring areas while ROD_RADIUS is symmetric changing the dome effect middle vs. at the end.

For eeprom you can switch also to sd card solution. Only requires you have a sd card in the printer. We use that for boards without eeprom. In pins.h for that board set
#define EEPROM_AVAILABLE EEPROM_SDCARD
to tell it to use sd card instead. Do NOT change it in configuration.h - will not work.

mj996, 2020-05-26:
> Actually it is ROD_RADIUS that gets used for computation. In the original config it get scomputed by subtracting CARRIAGE_HORIZONTAL_OFFSET and END_EFFECTOR_HORIZONTAL_OFFSET from PRINTER_RADIUS but [...]

OK I inserted a sd card and put #define EEPROM_AVAILABLE EEPROM_SDCARD in board section of pins.h.
Now I should put the number of EEPROM mode on 3  in  configuration.h file,right?
I did that and now I don't get '' No start signal detected '' anymore but each time that I send m502 & m500 commands in the host I get "could not write eeprom to sd card " error.What did I miss?
I put pins.h and host log files in order :

// RADDS Board
// http://www.dr-henschke.de/RADDS_due.html
#if MOTHERBOARD == 402
#ifndef __SAM3X8E__
#error Oops!  Make sure you have 'Arduino Due' selected from the 'Tools -> Boards' menu.
#endif
#define KNOWN_BOARD
#define CPU_ARCH ARCH_ARM
/*****************************************************************
 * Arduino Due Pin Assignments
 ******************************************************************/

#define ORIG_X_STEP_PIN 24
#define ORIG_X_DIR_PIN 23
#define ORIG_X_MIN_PIN 28
#define ORIG_X_MAX_PIN 34
#define ORIG_X_ENABLE_PIN 26

#define ORIG_Y_STEP_PIN 17
#define ORIG_Y_DIR_PIN 16
#define ORIG_Y_MIN_PIN 30
#define ORIG_Y_MAX_PIN 36
#define ORIG_Y_ENABLE_PIN 22

#define ORIG_Z_STEP_PIN 2
#define ORIG_Z_DIR_PIN 3
#define ORIG_Z_MIN_PIN 32
#define ORIG_Z_MAX_PIN 38
#define ORIG_Z_ENABLE_PIN 15

// Note that on the Due pin A0 on the board is channel 2 on the ARM chip
#define HEATER_0_PIN 13
// Due analog pin #54
#define TEMP_0_PIN 7
#define HEATER_1_PIN 7
#define TEMP_1_PIN 3
// Due analog pin #58
#define HEATER_2_PIN 12
// Due analog pin #55
#define TEMP_2_PIN 6
#define HEATER_3_PIN 11
// Due analog pin #56
#define TEMP_3_PIN 5
// Due analog pin #57
#define TEMP_4_PIN 4

// Due analog pin #59 = A5 -> AD 2
#define THERMOCOUPLE_0_PIN 2
// There are no more analog pins freely available.
// You can use direction and enable pin from extruder 0 socket as they are also
// analog pins. Then you need to move the stepper driver to a different socket.

// Direction pin of extruder 0
#define THERMOCOUPLE_1_PIN 1
// Step pin of extruder 0
#define THERMOCOUPLE_2_PIN 0
// Enable pin of extruder 0
#define THERMOCOUPLE_3_PIN 10

#define ORIG_E0_STEP_PIN 61
#define ORIG_E0_DIR_PIN 60
#define ORIG_E0_ENABLE_PIN 62

[... 117 lines cut ...]
15:08:50.864 : wait
15:08:51.849 : Error:Could not write eeprom to sd card
15:08:51.864 : T:26.25 /0 B:11.40 /0 B@:0 @:0
15:08:51.864 : wait
15:08:52.864 : wait
15:08:52.880 : T:26.25 /0 B:11.40 /0 B@:0 @:0
15:08:53.864 : wait
15:08:53.911 : T:26.25 /0 B:11.60 /0 B@:0 @:0
15:08:54.864 : wait
15:08:54.926 : T:26.25 /0 B:11.60 /0 B@:0 @:0
15:08:55.863 : wait
15:08:55.957 : T:26.25 /0 B:11.60 /0 B@:0 @:0
15:08:56.863 : wait
15:08:56.972 : T:26.25 /0 B:11.60 /0 B@:0 @:0
15:08:57.863 : wait
15:08:58.003 : T:26.25 /0 B:11.60 /0 B@:0 @:0
15:08:58.863 : wait
15:08:59.018 : T:26.25 /0 B:11.40 /0 B@:0 @:0
15:08:59.862 : wait
15:09:00.049 : T:26.25 /0 B:11.40 /0 B@:0 @:0
15:09:00.878 : wait
15:09:01.073 : T:26.25 /0 B:11.60 /0 B@:0 @:0
15:09:01.871 : wait
15:09:02.089 : T:26.25 /0 B:11.60 /0 B@:0 @:0
15:09:02.870 : wait

mj996, 2020-05-26:
Update : There's no matter which mode of EEPROM I choose,it will show  ''Error:Could not write eeprom to sd card '' after few seconds I connect the printer.

repetier, 2020-05-26:
The problem is not your eeprom setting. You seem to be not able to use the sd card, so check if you can read the sd card, e.g. list files using M20. When you can read files I'm quite sure eeprom to sd card will also start to work. If you have no feature controller selected it will use the sd card reader on the radds board, so try that first. Also make sure sd card is using a fat or vfat file system.

mj996, 2020-05-28:
> The problem is not your eeprom setting. You seem to be not able to use the sd card, so check if you can read the sd card, e.g. list files using M20. When you can read files I'm quite sure eeprom to [...]

Ok I guess the EEPROM is Ok because I can read the files on sd card with M20 command.
So now that I have eeprom why not trying to enable Z probe.
I mounted a micro switch on the effector as my Z probe .My Z_probe is Connected to the Z_min_Endstop. The Configuration of the Firmware seems to be ok: When I send the M119, I got the Message back: Z-Probe state:L. By triggering the Probe by hand and resending M119 I get: Z-Probe state:H. Then I send the G28 -> My Delta homes all three axes, and stops properly at the top of the extrusion. Then I send the G32 S2 ,The Effector begins to go down in direction of the build-platform.It hit the bed triggered the probe but then it stopped and showed the fatal error.
I homed the printer and send M119 and the Z-probe status was still H  !!
I think the problem is that the probe stays triggered after first hit but I don't know where does this problem come from !

Here are Z probe settings in config file : 
// #################### Z-Probing #####################

#define Z_PROBE_Z_OFFSET 0
#define Z_PROBE_Z_OFFSET_MODE 0
#define UI_BED_COATING 1
#define FEATURE_Z_PROBE 1
#define EXTRUDER_IS_Z_PROBE 0
#define Z_PROBE_DISABLE_HEATERS 0
#define Z_PROBE_BED_DISTANCE 10
#define Z_PROBE_PIN ORIG_Z_MIN_PIN
#define Z_PROBE_PULLUP 0
#define Z_PROBE_ON_HIGH 0
#define Z_PROBE_X_OFFSET 0
#define Z_PROBE_Y_OFFSET 0
#define Z_PROBE_WAIT_BEFORE_TEST 0
#define Z_PROBE_SPEED 2
#define Z_PROBE_XY_SPEED 150
#define Z_PROBE_SWITCHING_DISTANCE 1
#define Z_PROBE_REPETITIONS 1
#define Z_PROBE_USE_MEDIAN 1
#define Z_PROBE_HEIGHT 6.08
#define Z_PROBE_DELAY 0
#define Z_PROBE_START_SCRIPT ""
#define Z_PROBE_FINISHED_SCRIPT ""
#define Z_PROBE_RUN_AFTER_EVERY_PROBE ""
#define Z_PROBE_REQUIRES_HEATING 0
#define Z_PROBE_MIN_TEMPERATURE 150
#define FEATURE_AUTOLEVEL 0
#define FEATURE_SOFTWARE_LEVELING 0
#define Z_PROBE_X1 -77.22
#define Z_PROBE_Y1 -55
#define Z_PROBE_X2 77.22
#define Z_PROBE_Y2 -55
#define Z_PROBE_X3 0
#define Z_PROBE_Y3 104
#define BED_LEVELING_METHOD 1
#define BED_CORRECTION_METHOD 0
#define BED_LEVELING_GRID_SIZE 5
#define BED_LEVELING_REPETITIONS 5
#define BED_MOTOR_1_X 0
#define BED_MOTOR_1_Y 0
#define BED_MOTOR_2_X 200
#define BED_MOTOR_2_Y 0
#define BED_MOTOR_3_X 100
#define BED_MOTOR_3_Y 200
#define BENDING_CORRECTION_A 0
#define BENDING_CORRECTION_B 0
#define BENDING_CORRECTION_C 0
#define FEATURE_AXISCOMP 0
#define AXISCOMP_TANXY 0
#define AXISCOMP_TANYZ 0
#define AXISCOMP_TANXZ 0

PS : I'm using v1.0.4 dev of  firmware.

repetier, 2020-05-29:
With a mechanical endstop you need a pullup resistor. So set
#define Z_PROBE_PULLUP 1

or output is not defined and rather random.

mj996, 2020-05-29:
> With a mechanical endstop you need a pullup resistor. So set
> #define Z_PROBE_PULLUP 1
> 
> or output is not defined and rather random.

Ok now it works ,thanks.
So this is the process : 
G28 
G32 S2 (which measures 3 points)
G1 Z10
bed height map tool and measuring heights
G33 for Z correction
When I reach to this level and I command G33 , the printer starts measuring the points but at the middle of this procedure a fatal error comes from nowhere and the Z  correction procedure stops.

Here is the result of my bed height map measurement : 

X:	      -80.00	-53.33	-26.67	0.00	        26.67	53.33	80.00
y:   80.00	14.44	14.84	15.22	15.62	16.12	16.65	17.24
y:   53.33	14.52	14.68	15.22	15.49	15.84	16.27	16.79
y:   26.67	14.43	14.81	15.04	15.32	15.67	16.09	16.54
y:   0.00	14.52	14.75	14.96	15.26	15.50	15.82	16.19
y:  -26.67	14.40	14.46	14.92	15.17	15.34	15.59	15.77
y:  -53.33	14.58	14.61	14.74	15.01	15.05	15.14	15.34
y:  -80.00	14.75	14.85	15.00	14.93	14.70	14.78	14.98

and here are my Z correction settings in config file : 

#define DISTORTION_CORRECTION 1
#define DISTORTION_CORRECTION_POINTS 7
#define DISTORTION_LIMIT_TO 2
#define DISTORTION_CORRECTION_R 100
#define DISTORTION_PERMANENT 1
#define DISTORTION_UPDATE_FREQUENCY 15
#define DISTORTION_START_DEGRADE 0.5
#define DISTORTION_END_HEIGHT 1
#define DISTORTION_EXTRAPOLATE_CORNERS 0
#define DISTORTION_XMIN 10
#define DISTORTION_YMIN 10
#define DISTORTION_XMAX 190
#define DISTORTION_YMAX 190

mj996, 2020-05-29:
So here's an update : 
I changed #define DISTORTION_EXTRAPOLATE_CORNERS to 1
and
#define DISTORTION_XMIN 10 to -100
#define DISTORTION_YMIN 10 to -100
#define DISTORTION_XMAX 190 to 100
#define DISTORTION_YMAX 190  to 100
and now I am able to run G33 completely but I think I still miss something because I can't get to the fine results.

After G33 command I opened bed height map tool and measured  heights again but actually there was no obvious difference.
I should mention that each time I want to send G32 or G33 commands I bring the nozzle to 15mm above the bed.(My Z probe height is about 9 mm)

Maybe I'm doing something wrong or I miss some parameters in firmware.
Here are the results before Z correction : 

X:	       -70.00	-46.67	-23.33	0.00	        23.33	46.67	70.00
y:   70.00	14.54	14.93	15.29	15.63	16.04	16.52	17.12
y:   46.67	14.49	14.69	15.24	15.61	15.98	16.35	16.78
y:   23.33	14.37	14.70	15.11	15.37	15.70	16.02	16.45
y:   0.00	14.39	14.56	14.94	15.28	15.53	15.77	16.12
y:  -23.33	14.54	14.43	14.57	15.08	15.33	15.62	15.91
y:  -46.67	14.66	14.39	14.36	14.79	15.11	15.29	15.51
y:  -70.00	14.85	14.47	14.39	14.79	15.03	15.11	15.21

Here is the distortion matrix : 

118, 22, -42, -93, -160, -247, -346
 101, 59, 13, -63, -136, -211, -297
 52, 40, -15, -62, -110, -173, -235
 91, 71, 54, -23, -72, -120, -173
 63, 75, 3, -50, -59, -84, -123
 49, 103, 100, 35, -33, -53, -73
 73, 112, 112, 41, -13, -25, -30

And here are results after Z correction when I remeasured heights in bed height map tool :

X:	       -70.00	-46.67	-23.33	 0.00	23.33	46.67	70.00
y:   70.00	14.49	14.87	15.26	15.66	16.11	16.60	17.13
y:   46.67	14.49	14.76	15.14	15.53	15.88	16.31	16.75
y:   23.33	14.47	14.75	15.05	15.31	15.68	16.09	16.55
y:   0.00	14.44	14.71	14.94	15.25	15.47	15.82	16.22
y:  -23.33	14.49	14.50	14.82	15.14	15.41	15.56	15.89
y:  -46.67	14.62	14.29	14.37	15.06	15.26	15.34	15.59
y:  -70.00	14.78	14.34	14.34	14.76	15.06	15.21	15.29
 
and here are some settings of my config file.Maybe I missed something.

#define DISTORTION_CORRECTION 1
#define DISTORTION_CORRECTION_POINTS 7
#define DISTORTION_LIMIT_TO 2
#define DISTORTION_CORRECTION_R 70
#define DISTORTION_PERMANENT 1
#define DISTORTION_UPDATE_FREQUENCY 15
#define DISTORTION_START_DEGRADE 0.5
#define DISTORTION_END_HEIGHT 1
#define DISTORTION_EXTRAPOLATE_CORNERS 1
#define DISTORTION_XMIN -100
#define DISTORTION_YMIN -100
#define DISTORTION_XMAX 100
#define DISTORTION_YMAX 100

// ##########################################################################################
// ##                           Movement settings                                          ##
// ##########################################################################################

#define FEATURE_BABYSTEPPING 0
#define BABYSTEP_MULTIPLICATOR 1

#define DELTA_SEGMENTS_PER_SECOND_PRINT 180 // Move accurate setting for print moves
#define DELTA_SEGMENTS_PER_SECOND_MOVE 70 // Less accurate setting for other moves
[... 190 lines cut ...]
#define LANGUAGE_FR_ACTIVE 1
#define LANGUAGE_CZ_ACTIVE 0
#define LANGUAGE_PL_ACTIVE 1
#define LANGUAGE_TR_ACTIVE 1
#define LANGUAGE_RU_ACTIVE 0
#define UI_PRINTER_NAME "RepRap"
#define UI_PRINTER_COMPANY "Home made"
#define UI_PAGES_DURATION 4000
#define UI_SPEEDDEPENDENT_POSITIONING 0
#define UI_DISABLE_AUTO_PAGESWITCH 1
#define UI_AUTORETURN_TO_MENU_AFTER 30000
#define FEATURE_UI_KEYS 0
#define UI_ENCODER_SPEED 1
#define UI_REVERSE_ENCODER 0
#define UI_KEY_BOUNCETIME 10
#define UI_KEY_FIRST_REPEAT 500
#define UI_KEY_REDUCE_REPEAT 50
#define UI_KEY_MIN_REPEAT 50
#define FEATURE_BEEPER 0
#define CASE_LIGHTS_PIN -1
#define CASE_LIGHT_DEFAULT_ON 1
#define UI_START_SCREEN_DELAY 1000
#define UI_DYNAMIC_ENCODER_SPEED 1

I'm so thankful for your supports.

[46 more replies]