repetier/RepetierFirmware #924: Help Please

tsvikabp, 2020-04-22:
Hi,
After spending hours trying solving this - your help is much appreciated.
My homemade CNC is up and running using Arduino mega + RAMPS 1.4.
All is working great when using Gcode.
however, when using keys :
UI_KEYS_BUTTON_LOW(31,1021);
 UI_KEYS_BUTTON_LOW(33,1024); 
 UI_KEYS_BUTTON_LOW(35,104); 
 UI_KEYS_BUTTON_LOW(37,105); 
 UI_KEYS_BUTTON_LOW(39,1023); 
 UI_KEYS_BUTTON_LOW(41,102); 
 UI_KEYS_BUTTON_LOW(43,103); 
 UI_KEYS_BUTTON_LOW(45,1022); 
 UI_KEYS_BUTTON_LOW(47,100); 
 UI_KEYS_BUTTON_LOW(32,101); 
  'Home' and 'All home' are moving OK
but x,y,z axis are moving VERY slowly using those keys (Buy OK using Gcode) - whatever I tried to change - nothing helps.
*****
I have another issue  is with the display  (20X4)
all I need is to see is the current position of the axis (no menu whatsoever - directly to the numbers) 
currently, all I see is:
>English
****
can you please help me?
Thank you

repetier, 2020-04-23:
In the display it asks for language for interface. Press ok button to set language. If have none assigned set language id in eeprom. 0 is first language compiled - so if you only have english 0 is the value.

Actions 100-105 move in 1mm steps:
       case UI_ACTION_X_UP:
        case UI_ACTION_X_DOWN:
            if (!allowMoves)
                return action;
            PrintLine::moveRelativeDistanceInStepsReal(((action == UI_ACTION_X_UP) ? 1.0 : -1.0) * Printer::axisStepsPerMM[X_AXIS], 0, 0, 0, Printer::homingFeedrate[X_AXIS], false, false);
            break;

if they do not move 1mm per click your steps per mm is configured wrong.
Normally you use an encoder for moving xyz which is much faster.

You might speed up repetition when holding it down:
/** \brief First time in ms until repeat of action. */
#define UI_KEY_FIRST_REPEAT 500
/** \brief Reduction of repeat time until next execution. */
#define UI_KEY_REDUCE_REPEAT 50
/** \brief Lowest repeat time. */
#define UI_KEY_MIN_REPEAT 50

As you see first repeat is after a half second and each repeat gets 50ms faster until you are down to 50ms per repeat which is 20mm/s.

tsvikabp, 2020-04-23:
Thank You ! as you recommend, I'd install a rotary encoder - Life is much better now !!!