repetier/RepetierFirmware #925: M3/M4/M5 D9 config  Problem

tsvikabp, 2020-05-04:
I connected 12v relay to D9 in order to start up the CNC spindle motor.
but all I get is very short blink (led) - the D9 needs to be HIGH , until M5 code sent and not PWM
the config file :

#define NUM_EXTRUDER 0
#define MOTHERBOARD 33
#include "pins.h"
***
#undef FAN_BOARD_PIN
#define FAN_BOARD_PIN -1
#define BOARD_FAN_SPEED 255
#define BOARD_FAN_MIN_SPEED 0
#define FAN_THERMO_PIN -1
#define FAN_THERMO_MIN_PWM 128
#define FAN_THERMO_MAX_PWM 255
#define FAN_THERMO_MIN_TEMP 45
#define FAN_THERMO_MAX_TEMP 60
#define FAN_THERMO_THERMISTOR_PIN -1
#define FAN_THERMO_THERMISTOR_TYPE 0
***
#define SUPPORT_CNC 1
#define CNC_WAIT_ON_ENABLE 300
#define CNC_WAIT_ON_DISABLE 0
#define CNC_ENABLE_PIN ORIG_FAN_PIN
#define CNC_ENABLE_WITH 1
#define CNC_DIRECTION_PIN -1
#define CNC_DIRECTION_CW 1
#define CNC_PWM_MAX 255
#define CNC_RPM_MAX 10000
#define CNC_SAFE_Z 150
#define FEATURE_FAN_CONTROL 1

HEATER_2_PIN - not use.

I vaguely remember that last time I hade this problem (before the upgrade) I changed some settings in the config.h (maybe PWM from 0 to 255 somewhere ) making D9 constant HIGH when using M3 or M4. but not sure where, and now I can't find it... is the NUM_EXTRUDER 0 should be set to NUM_EXTRUDER 1? 
Thank you.

repetier, 2020-05-05:
spindleOff is only called in M5 command. Problem is I think
#define FEATURE_FAN_CONTROL 1

since fan uses same pin and then fan controls the output not M3/M5. So either set fan control 0 or define a different pin for the fan if you have one.

tsvikabp, 2020-05-06:
Opps.....Thx