repetier/RepetierFirmware #926: Dev2: Serial port USB issue

JeppeIbsen, 2020-05-07:
I'm testing the dev2 repetier on one of our custom cartesian printers, with bare due electronics connected to octoprint on a pi. 

Unfortunately when the v2 firmware is uploaded to the due, there is no longer any open serial ports and it's not possible to connect. I have to upload a blink sketch to the programming port to reach the due again. I have been using the v1 repetier on different printers throughout the years, but the configuration on the v2 is somehow wrong for my setup.

Configration h:
https://pastebin.com/fLPndPRE

Radds.h where the pins are defined:
https://pastebin.com/arhDfULH

platformio.ini:
https://pastebin.com/jwfcHy6P

How do I correctly configure the v2 firmware, so I can connect to it via USB ?

repetier, 2020-05-07:
> bare due electronics
So no RADDS board as you defined it?

Which usb port do you want to use. You have 
#define RFSERIAL SerialUSB
which is the native usb port. For programming port use Serial or to use both set
#define BLUETOOTH_SERIAL Serial

If it is no radds and you have no eeprom add in configuration.h
#define EEPROM_AVAILABLE EEPROM_NONE

If these are all changes to default cartesian config it should compile and already connect. If not best is to go back and add changes incrementally when a first version worked, so you know wher ethe problem arises.

I also saw you did not mention configuration_io.h which is in V2 an important part of the configuration!

JeppeIbsen, 2020-05-10:
Unfortunately even when defining eeprom_none, there is still no USB access to the arduino. Windows claims that 'USB device not recognized' every 30 seconds, it might be resetting.

Do you have a 'working' image of the firmware, for a bare-due ? I can then work forward from that.
I looked into the configuration_io and it seemed OK, however we don't use TMC2130 drivers, but I dont now what an alternative confugration in the file would look like.

The configuration_io.h is here:
https://pastebin.com/LqLqLKcc

repetier, 2020-05-11:
Check cartesian/felix pro for regular stepper drivers. They just use 
STEPPER_SIMPLE(name, stepPin, dirPin, enablePin, minEndstop, maxEndstop) 
https://docfirmwarev2.repetier.com/config/motors#stepper_simple

instead of the trinamic. Might also be the source if spi does not work on set pins.

I have no bare due sample. Default config with radds and eeprom mode 0 should work on bare due also it will freak out from defect thermistor but should give some lines on connect with reset over programming port which it uses.