repetier/RepetierFirmware #933: Xyz serial number... For those already switched over

jack30977, 2020-06-09:
So since you guys converted your 3d printers to repetier and I am a dumbnut who can't figure this one out. I converted back to xyz and I am asking if I could have anyone's serial number so that way I can use it to register my own because I picked mine up from a dumpster. Therefor I do not have access to the back panel.
Thanks if you can give me your serial number,

MasterCATZ, 2020-07-21:
do you still need serial ? I lent my Da Vinci 2.0 Duo out yesterday but can ask them to get the number for me , tho I do have its box here still that might have it

jack30977, 2020-07-21:
Yeah I still need a serial number. Its located on the back cover. Thank you for responding back to me