repetier/RepetierFirmware #934: FEATURE REQUEST : allow to get status of command instead of set default value

luc-github, 2020-06-27:
Hello,
I am working on ESP3D and wifi frontend and today it is not easy to get the value of FeedRate / FlowRate on Repetier 
When on Smoothieware / Marlin 
* Sending M220 give current Feedrate, in Repetier if set feedrate to 100%
* Sending M221 give current Flowrate, in Repetier if set flowrate to 100%

I know there is M408 but it is not always enabled in FW

Would it be possible to change current behavior? or may be add a syntax that allow to query the value without changing it ?
like `M220 S-1` and `M221 S-1` this would keep current backward compatibility ?

repetier, 2020-06-27:
That is normally not needed. The firmware already repeats changes on all serial connections like this:

FlowMultiply:89
SpeedMultiply:154

If you send empty M220/M221 at least with 1.0.4 and 2.0.x.

repetier, 2020-06-27:
Sorry, just saw that it returns a value but sets it to 100. Will change that for next update. But it will be a problem of backward compatibility. So you better assume 100 at startup and only watch for change messages.

luc-github, 2020-06-27:
>That is normally not needed. The firmware already repeats changes on all serial connections like this:

>FlowMultiply:89
>SpeedMultiply:154

Yes but when WebUI connect the serial connection is already established with ESP Firmware so it won't be seen by WebUI

>Sorry, just saw that it returns a value but sets it to 100. Will change that for next update. But it will be a problem of backward compatibility

Yes that was my guess , what about my proposal to use a M220 S-1 M221 S-1 to send the status instead of changing it ?
so no change if M220 / M221

repetier, 2020-06-27:
> Yes but when WebUI connect the serial connection is already established with ESP Firmware so it won't be seen by WebUI
What is the difference between WebUI and ESP Firmware? I thought it is the same software and which one talks to firmware? As long as they init more or less together with printer firmware that should be no problem. The part of your software doing firmware communication is responsible for storing value internally so that should be able to give latest value to ui. I mean you should not ask firmware any ui update for values. That would block too much cpu on firmware. That is why we use the push data on change solution after all. Also I admit having a way to pull it would be nice.

>Yes that was my guess , what about my proposal to use a M220 S-1 M221 S-1 to send the status instead of changing it ?
so no change if M220 / M221
That is no solution. Using illegal values with special meanings is bad. And you will always have the backward compatibility problem. Old firmwares will set value to 25 instead as they clip values instead of ignoring illegal ones. I even can't add a new parameter R for report since old firmware would still set it to 100%,

luc-github, 2020-06-27:
ESP FW is just bridge wifi to serial, not storing any data - when WEBUI is a javascript /html sending commands
Also  changing FeedRate from printer when WebUI is not connected has same problem, webUI won't see the changes and assuming 100% is not working then

The backward compatibility concern was for senders - not old FW, as solution is in new FW I think - I wanted to do PR to Repetier and update my fork accordingly our discussion but seems no solution if no additionnal parameter cannot be considered neither
 
I just wanted to raise the problem as Bigtreetech TFT works in same way and they may not be the only one - I still can have the M408 as backup solution but M220/M221 was more  consistent IMHO

repetier, 2020-06-27:
> ESP FW is just bridge wifi to serial, not storing any data - when WEBUI is a javascript /html page sending commands
Ok, understand. So not like our server but with a "stupid" connector and all the logic on the javascript side once user calls the page.
> Also changing FeedRate from printer when WebUI is not connected has same problem, webUI won't see the changes and assuming 100% is not working then

Yes, it would show false value until it is changed again.
> 
> The backward compatibility concern was for senders - not old FW, as solution is in new FW I think - I wanted to do PR to Repetier and update my fork accordingly our discussion but seems no solution if no additionnal parameter cannot be considered neither
Next version 1.0.5 and next update on 2.0.x will contain the fix at least.
> 
> I just wanted to raise the problem as Bigtreetech TFT works in same way and they may not be the only one - I still can have the M408 as backup solution but M220/M221 was more consistent IMHO
Haven't tested the Bigtreetech solution so far. But have it laying here already for testing.

luc-github, 2020-06-27:
>Next version 1.0.5 and next update on 2.0.x will contain the fix at least

ho that is perfect thank you ^_^

luc-github, 2020-06-27:
I guess issue can be closed now - thank you

closed 2020-06-27 by luc-github