repetier/RepetierFirmware #938: Homing all axes does not execute Z-homing

walterbux, 2020-07-06:
Hello,
I have just tested the new firmware with an old configuration.h perfectly working in an earlier issue (1.03)..
What happens is that by sending the G28 command the printer executes X homing, Y homing and when executing Z homing it stops while it should go down toward the bed for z-probing.

Best regards.

Walter

p.s. I add the configuration.h used
[Configuration.zip](https://github.com/repetier/Repetier-Firmware/files/4879968/Configuration.zip)

repetier, 2020-07-06:
When doe sit stop - directly or does it move a while. Also on retest does it move deeper? If so it would be most likely the z endstop triggering early.

walterbux, 2020-07-07:
After X and Y homing it moves a little up and a little down but it stops Z motion and continues x and y positioning as if it had reached Z=0. On retest it moves in the same way as the first time.

I underline that I use the same HW and SW configuration of 1.03 release. By flashing this release everything works perfectly therefore my deduction is that either is a configuration error or a bug in the new release.

repetier, 2020-07-07:
I understood that you use same firmware, but on the other side I have no problems with it and also no other users with that problem. That is why I suspected a hardware problem - crosstalk to z endstop - which causes exactly your pattern.

Can you try homing Z with
M111 S70
M119
G28 Z0

and report the output.

The S70 enables debugging endstops so we can see if firmware thinks endstop is triggered. 

An other reason might be that eeprom has bad values. There are good chances I changed eeprom layout a bit so it might now contain wrong values like wrong z height, steps per mm, ... 

Quite sure it is one of these two.

walterbux, 2020-07-07:
Hello,
meantime you were writing I was making additional testing that in part includes what you are asking me to do.

I have reflashed the new FW. I have noted that sending commands from Repetier Host I continuously receive Unknown command or checksum error. I use the bluetooth (wifi) interface and I connect in TCP/IP with repetier protocol and ping-pong communication. The same that I do with release 1.0.3 but in this case neither unknown commands nor checksum errors are present. I have tested the endstops with the M119 command and found correct behaviour of them finding the answer inside the list of checksum errors.
I confirm the behaviour of the Homing that this time I issued from the display menu.
X homing first, Y homing after then Z up a little, down a little then X=0, Y=0 position movement. The bed is far from Zprobe=Zmin inductive sensor.

Then retested old release 1.0.3 and perfect behaviuor no error nothing wrong.
Regards.

Walter

walterbux, 2020-07-07:
When I flashed the new FW I issued a reset Eprom by front panel then I restarted the firmware by power cycling the machine.
But I have also made another test: I tryed homing by flashing the new FW with eprom disabled ... same homing behaviour.