repetier/RepetierFirmware #957: Dev2: Rumba32 MKS board compatibility

GenaUser, 2020-08-31:
Hello,
Are you planning to add compatibility with "Rumba32 MKS" board in the near future?
This would allow the use of a TFT screen, as the classic Rumba32 does not have a dedicated RX / TX connector.
Thank you in advance for your information.

repetier, 2020-08-31:
Rimba32 is already supported. Only issue is that the new stm32 library 8..0.0 introduced incompatible changes. Using the old 6.1.1 version allows compilation until I have fixed the problem.

GenaUser, 2020-09-01:
Thank you.
I manage to compile (on windows, not on MacOS) and download the "Rumba32 Aus3D" environment on my "Rumba32 MKS" (with TFT socket), but the communication with a TFT screen does not seem to work.
Will it be possible to add the Rumba32_MKS option in the firmware, the libraries exist for this specific card?
The Rumba32 MKS is less expensive than the Aus3D and offers connectivity with TFT screens.

repetier, 2020-09-01:
Aren't they the same boards from different producers?

Anyway, the displays use serial right? Don't have them so can not really say much about them. But in that case it matters how the display firmware is written and if it is also compatible with repetier V2 firmware. If they support the firmware you would just define a second serial in firmware and connect the display.

GenaUser, 2020-09-02:
They are not quite the same boards
[image]
https://github.com/makerbase-mks/MKS-RUMBA32
https://github.com/Aus3D/RUMBA32
[image]

I will try to define a second serial as you indicate
Thank you for your job and your advice.

GenaUser, 2020-09-04:
I'm sorry, I didn't find how define a new serial on PA9 and PA10 pin in dev2

AbsoluteCatalyst, 2020-09-06:
I don't have that board, however; in your configuration.h file set BLUETOOTH_SERIAL -1 to BLUETOOTH_SERIAL **1**, and in platformio.ini add an extra build flag: HAVE_HWSERIAL1 ("**-DHAVE_HWSERIAL1**") 
This should enable USART1 @ 115200 default baud (or whatever you have in your EEPROM). Just connect and go.

Unfortunately your TFT may or may not really be compatible with repetier FW as stated above. 
I have a bigtreetech TFT 3.5 that functions for most controls, however it isn't able to read my onboard SDCard.

(I'm assuming it's waiting for a very marlin-like firmware response to a M20 command or something, but I haven't diagnosed it as I only received it last night.) 

Worth a try!

GenaUser, 2020-09-07:
@AbsoluteCatalyst, thanks a lot for your help.
Now, the Bigtreetech TFT 3.5 display connects to the Rumba32 MKS board on the dedicated AUX-1 pin.
There is now a long way to go in trying to configure the firmware.
Thanks again

repetier, 2020-09-07:
I have added the description to the rumba32.h board file. Closing the issue a sit seems solved.

closed 2020-09-07 by repetier