repetier/RepetierFirmware #963: [feature request]  configuring STOP button to lift the hot end and turn off the cooling fan

ahmedsohair, 2020-09-07:
So i recently configured my 3D printer using the config tool for Repetier firmware. However, there's a little change that i need to make when i press the STOP button. What i want is that when i press the STOP button, the hot end should be lifted and the the cooling fan should be turned off. I am unable to figure this using the config tool. I have a slight idea that this might be possible by changing the functions already being used but don't know where to start from and which files or which lines of code to edit. Any help would be appreciated

repetier, 2020-09-08:
Guess you mean when sd printing. There is no config option to change behaviour but you can modify
SDCard::stopPrint() 

to also send additional commands on stop.
GCode::executeFString(PSTR("M107\nG91\nG1 Z10 F800\nG90"));

or what ever you want. Just put it at the end of the function in sdcard.cpp

ahmedsohair, 2020-09-08:
Thanks a lot. 
I'll check this one and update.

Currently i was thinking of adding something like this in the stopPrint() func :

Commands::setFanSpeed(0, true);
Printer::homeAxis(false, false, true);