repetier/RepetierFirmware #984: dev2: Homing not correct if light barrier is alrady covered

JoeScotty, 2020-10-04:
Being at first tests with dev2 at my cartesian printer with BLTouch,
there seem to be a homing problem:

If the X or Y homing light barrier is covered when HomeAll is started,
the X-motion is in positive direction, which is correct.
But it stops at the light barrier switching position,
while it should stop at the position "X axis endstop distance after homing".
There is also no retest.

Found this problem while playing with negative offset values.

Propose the following steps if homing is started and the Light barrier/sensor is already covered/active:
-move opposite to homing direction until sensor is free
-move further backwards by ENDSTOP_X_BACK_MOVE
-move to light sensor until triggered
-move to ENDSTOP_X_BACK_ON_HOME

The homing & probing parameters are as primitive as possible in these first tests:

configuration.h
#define X_HOME_PRIORITY 0
#define Y_HOME_PRIORITY 1
#define Z_HOME_PRIORITY 2

#define X_HOME_DIR -1
#define Y_HOME_DIR -1
#define Z_HOME_DIR -1

#define ENDSTOP_X_BACK_ON_HOME 0.5  // EEPROM "X axis endstop distance after homing" 0.5
#define ENDSTOP_Y_BACK_ON_HOME 0.5  // EEPROM "Y axis endstop distance after homing" 0.5
#define ENDSTOP_Z_BACK_ON_HOME 0    // EEPROM "Z axis endstop distance after homing" 0
#define ALWAYS_CHECK_ENDSTOPS 0
#define MOVE_X_WHEN_HOMED 0
#define MOVE_Y_WHEN_HOMED 0
#define MOVE_Z_WHEN_HOMED 0
#define HOMING_FEEDRATE_X 10       // EEPROM "X-axis homing feedrate" 10
#define HOMING_FEEDRATE_Y 10       // EEPROM "Y-axis homing feedrate" 10
#define HOMING_FEEDRATE_Z 10       // EEPROM "Z-axis homing feedrate" 10

#define ZHOME_PRE_RAISE 2
#define ZHOME_PRE_RAISE_DISTANCE 5
#define ZHOME_MIN_TEMPERATURE 0
#define ZHOME_HEAT_ALL 0
#define ZHOME_HEIGHT 4
#define FIXED_Z_HOME_POSITION 1 //still 1, because ZHOME_POS will move after fisrt homing tests
#define ZHOME_X_POS 0
#define ZHOME_Y_POS 0

#define ENDSTOP_X_BACK_MOVE 2      
#define ENDSTOP_Y_BACK_MOVE 2      
#define ENDSTOP_Z_BACK_MOVE 3
#define ENDSTOP_X_RETEST_REDUCTION_FACTOR 3
#define ENDSTOP_Y_RETEST_REDUCTION_FACTOR 3
#define ENDSTOP_Z_RETEST_REDUCTION_FACTOR 3

#define Z_PROBE_TYPE Z_PROBE_TYPE_BLTOUCH 
#define Z_PROBE_HEIGHT 2             // EEPROM "Trigger Height" 2
#define Z_PROBE_BED_DISTANCE 10      // EEPROM "Min Nozzle Distance" 10
#define Z_PROBE_SPEED 5              // EEPROM "Z-Probe probing speed 5
#define Z_PROBE_X_OFFSET 0           // EEPROM "Z-Probe Y offset" 0
#define Z_PROBE_Y_OFFSET 0           // EEPROM "Z-Probe Y offset" 0
#define Z_PROBE_COATING 0
#define Z_PROBE_DELAY 1000           // EEPROM "Z-Probe Deploy Delay"
#define Z_PROBE_REPETITIONS 1
#define Z_PROBE_USE_MEDIAN 1 
#define Z_PROBE_SWITCHING_DISTANCE 2
#define Z_PROBE_BORDER 0
#define Z_PROBE_PAUSE_HEATERS 0
#define Z_PROBE_START_SCRIPT ""
#define Z_PROBE_FINISHED_SCRIPT ""
#define Z_PROBE_RUN_AFTER_EVERY_PROBE ""
#define Z_PROBE_PAUSE_HEATERS 0      // EEPROM "Z-probe Pause heaters"
#define Z_PROBE_TEMPERATURE 0

repetier, 2020-10-04:
Not sure if you found a bug or if it was caused by start positions. First tool should not really be behind switching position. With mechanical end stops this is of course more a problem than with optical where you might have more room. In any case ENDSTOP_X_BACK_MOVE should be choosen to be higher then the maximum distance that is physically possible. So when end stop is already triggered it will stop first move directly and move back to untrigger and then do second test.

Now why I'm not sure. If you stared around 2.5mm behind sensor you would just see the 2mm and 0.5mm move in positive direction which might end just close behind the detector. So as a counter test what happens with 20 and 10 mm instead? Should be better visible and in any case move it outside the light barrier. If it still fails it is a bug otherwise just bad  ENDSTOP_X_BACK_MOVE choice.

JoeScotty, 2020-10-05:
Yes, the tool should not be behind the switching position. It just was a test for behaviour in unusual condition,
which I got when playing with negative offset values.

You are right, the 2mm BACK_MOVE plus 0,5mm BACK_ON_HOME was an unlucky combination.
With 20mm plus 10mm it first moves back 20mm, then retest and then moves back 10mm to the final position.
So just the BACK_MOVE value must be bigger than the possible distance behind the switching position.

I was focused on first detecting the switching position also when moving backwards,
but it also works when following the above rule.
Thanks for your quick help!

closed 2020-10-05 by JoeScotty