repetier/RepetierFirmware #988: extruder does not move reptier v 1.0.4

qusai210, 2020-10-06:
i build the 1st 3d printer and the extruder does not move 
can you help me 
what is the reason

repetier, 2020-10-06:
Was you extruder hot while trying? Extruder will not work when cold except if you explicitly allow it setting allow cold extrusion with
M302 S1

Also check eeprom values in case first upload did not have set them as you want. Or restore to config settings with
M502
M500

Last thing is make sure dry run debug mode is not active - that also prevents extruder motion.

qusai210, 2020-10-06:
i move it manual but not move , and i trying when hot with m302 s1 and not move why?
where is the dry run , and how know is active or no ? 
 anther thing 
i used ramp shield 1.4 and did not use eeprom 
use the eeprom are important ?
my problem is still 
thank you

repetier, 2020-10-06:
You see it in the host console if printer switches into dry mode (happens when temp. sensors do not work). So check output in console/log. You can disable it with
M111 S6
Adding 8 to S enables it.
EEPROM or not is not relevant here.
When printer is just enabled or reset can you move extruder motor by hand? If not and it feels like motor is on you need to change
#define EXT0_ENABLE_ON 0
to 1. Some motor drivers enable power on 1 some on 0. Also swap with x motor if X works to rule out that driver is defect.

qusai210, 2020-10-06:
i swap with x motor and work but in extr0 no
and i change the   #define EXT0_ENABLE_ON 0
to
#define EXT0_ENABLE_ON 1
still the problem
what is the happen 
pls help me

qusai210, 2020-10-07:
?

qusai210, 2020-10-07:
i swap with x motor and no  work

repetier, 2020-10-07:
Are all drivers identical? Then set EXT0_ENABLE_ON on same value as X axis.

Connect motor to x axis motor to see if it moves there. Especially after you already swapped the drivers. That way you now motor and driver work.

Last test is to connect X motor to E0 and make config use the E0 socket for motion and set Extruder 0 to use X motor socket instead.

If then extruder moves and X motor not I would assume the socket is defect and you should switch to using E1 socket instead.

qusai210, 2020-10-07:
repetier the problem in  code 
all driver the same A8499
and i swap between  x motor and  extruder in config (just) ,,, 
when sweep motor  extruder work and x not work 
 notice i will use the standard config  in web site to fix the problem and used  v .92.9 but still the problem 
can you give me  config new to fix the problem or help me to fix them
thank you

repetier, 2020-10-07:
You can just try the default configuration in config tool just with your board selected.
Quite unlikely the both extruder ports are defect and with swapping you already tested that all other hardware is working as well.
If you also send M302 S1 before testing extruder I see no reason why it should not work from here. So please check if you really did all the things I said and all tests are positive. So a fresh version with basic settings for testing is the only thing to test I can think of.

qusai210, 2020-10-13:
hi repetier  , thenk you extruder is move when heater the nozzel or use the g code you give me 
but i have anther problem the z axis 
i set  step per mm , speed travel , acceleration 
when test z axis there is no change in them

why