repetier/RepetierFirmware #989: dev2: stepper inactivity issue

JoeScotty, 2020-10-09:
Cartesian printer:
Had some cases that after Power On and HomeAll the X/Y Position shown in the LCD Display was blinking with ????

Probable reason: At my earlier tests I set STEPPER_INACTIVE_TIME in EEPROM to 60000 (ms)
after adding debug information, I finally came to the code and added Debug Info
For tests, STEPPER_INACTIVE_TIME was set to 30000

`void Printer::defaultLoopActions() {
    Commands::checkForPeriodicalActions(true); //check heater every n milliseconds
    if (HAL::i2cError && HAL::i2cError != 255) {
        HAL::i2cError = 255; // Flag to show error message only once
        GCode::fatalError(Com::tI2CError);
    }
    millis_t curtime = HAL::timeInMilliseconds();
    if (isRescueRequired() || Motion1::length != 0 || isMenuMode(MENU_MODE_SD_PRINTING + MENU_MODE_PAUSED)) {
        previousMillisCmd = curtime;
    } else {
        curtime -= previousMillisCmd;
        if (maxInactiveTime != 0 && curtime > maxInactiveTime) {
            Printer::kill(false);
            //Debug Info
            Com::printFLN(PSTR(" Printer defaultloopaction curtime>maxinactivetime starts kill"));
        } else {
            Printer::setAllKilled(false); // prevent repeated kills
        }
        if (stepperInactiveTime != 0 && curtime > stepperInactiveTime) {
            Printer::kill(true);
            //Debug Info
            Com::printF(PSTR("Printer defaultloopaction curtime>stepperinactivetime starts kill curtime="), curtime);
            Com::printFLN(PSTR(" stepperInactiveTime="), stepperInactiveTime);
        }
    }
    `
    

Some Log examples (when XY Pos was blinking after HomeAll):
    20:36:11.044 : Printer defaultloopaction curtime>stepperinactivetime starts kill curtime=39895 stepperInactiveTime=30000

    20:37:04.074 : Printer defaultloopaction curtime>stepperinactivetime starts kill curtime=52378 stepperInactiveTime=30000

    20:39:49.581 : Printer defaultloopaction curtime>stepperinactivetime starts kill curtime=40395 stepperInactiveTime=30000

my Interpretation:
    curtime should be around 0 when homing motor activity just stopped.
    When havin STEPPER_INACTIVE_TIME of 600000 (10 Minutes), this would happen hardly, but it would happen.
    And hardly upcoming errors are sometimes very difficult to reproduce.
    In my mind the curtime should be reset to 0 when motor stopps after homing process.

repetier, 2020-10-09:
I think I see the problem. When you start after 20 seconds and homing takes more then 10 seconds it never adjusts previousMillisCmd so when homing is through and we start going back to defaultLoopActions it could time out. I have now added the same test in checkPeriodicalActions

```
void Commands::checkForPeriodicalActions(bool allowNewMoves) {
    Printer::handleInterruptEvent();
    FirmwareEvent::handleEvents();
#if EMERGENCY_PARSER
    GCodeSource::prefetchAll();
#endif
    if (Printer::isRescueRequired() || Motion1::length != 0 || Printer::isMenuMode(MENU_MODE_SD_PRINTING + MENU_MODE_PAUSED)) {
        previousMillisCmd = HAL::timeInMilliseconds();
    }
```

which gets called also in homing and other parts waiting for a condition. At least for homing that should solve the timeout problem.

repetier, 2020-10-09:
Closing as I think this fixes it. If not please feel free to reopen.

closed 2020-10-09 by repetier