repetier/RepetierFirmware #990: dev2: Offset handling issue

JoeScotty, 2020-10-09:
Cartesian printer. Having 2 Extruders and BLTouch with offset Values the planned setup will be

              BLTouch (-20,-40)
    
    E1(-40/0)             E0 (0/0)   
    
but during tests, where I played also with an offset for E0, there seem to be an issue:

as long as E0 offset = 0/0), activating tool1, tool2, an zProbe, the positioning is fine.

Test1
    Tool 1 offset  0/0
    Tool 2 offset  -10/-4
    Z-probe offset -15/-23

    Test steps were:
        activate E1 (which is tool1)
        HomeAll
        zProbe pos was 15/23 OK
        after homing pos 0/0 OK
        activate E2: moved to 10/4 OK
        HomeAll:
        zProbe pos 15/23 OK
        after homing 10/4 OK

Test2
    Tool 1 offset  -7/-3
    Tool 2 offset  -10/-4
    Z-probe offset -15/-23

    Test steps were:
        activate E1 (which is tool1)
        HomeAll
        zProbe pos was 8/20 NotOK should be 15/23
            probable calc X: -ProbeOffX + Tool1OffX (- (-15) + (-7) = 8)
            probable calc Y: -ProbeOffX + Tool1OffX (- (-23) + (-3) = 20)
        after homing pos 0/0 NotOK should be 7/3
            probable calc X: -Tool1OffX + Tool1OffX (- (-7) + (-7) = 0)
            probable calc Y: -ProbeOffY + Tool1OffY (- (-3) + (-3) = 0)
        activate E2: moved to 3/1 NotOK should be 10/4
            calc similar to above, Tool1Off are added while it should not been added.

How does it come?
    I added some debug information where you already proposed: 
```c++      
    void Motion1::setToolOffset(float ox, float oy, float oz) {
        //Debug Info
        Com::printF(PSTR(" setToolOffset: TOffXalt="), toolOffset[X_AXIS]);
        Com::printF(PSTR(" TOffYalt="), toolOffset[Y_AXIS]);
        Com::printF(PSTR(" OxNeu="), ox);
        Com::printF(PSTR(" OyNeu="), oy);

        if (Motion1::isAxisHomed(X_AXIS) && Motion1::isAxisHomed(Y_AXIS)) {
            //Debug Info
            Com::printF(PSTR("   XYHomed!"));

            setTmpPositionXYZ(currentPosition[X_AXIS] + ox - toolOffset[X_AXIS],
                              currentPosition[Y_AXIS] + oy - toolOffset[Y_AXIS],
                              currentPosition[Z_AXIS]);
            //Debug Info
            Com::printF(PSTR("   CurrPosX="), currentPosition[X_AXIS]);
            Com::printF(PSTR(" CurrPosY="), currentPosition[Y_AXIS]);
            Com::printF(PSTR(" TempPosX="), currentPosition[X_AXIS] + ox - toolOffset[X_AXIS]);
            Com::printF(PSTR(" TempPosY="), currentPosition[Y_AXIS] + oy - toolOffset[Y_AXIS]);

            moveByOfficial(tmpPosition, Motion1::moveFeedrate[X_AXIS], false);
            waitForEndOfMoves();
        } else {
            //Debug Info
            Com::printF(PSTR("   XYNotHomed!"));

            currentPosition[X_AXIS] += ox - toolOffset[X_AXIS];
            currentPosition[Y_AXIS] += oy - toolOffset[Y_AXIS];

            //Debug Info
            Com::printF(PSTR("   NewCurrPosX="), currentPosition[X_AXIS]);
            Com::printF(PSTR(" NewCurrPosY="), currentPosition[Y_AXIS]);

            updatePositionsFromCurrent();
        }
        toolOffset[X_AXIS] = ox;
        toolOffset[Y_AXIS] = oy;
                //Debug Info
        Com::printF(PSTR("     TOffXNeu="), toolOffset[X_AXIS]);
        Com::printFLN(PSTR(" TOffYNeu="), toolOffset[Y_AXIS]);

        updatePositionsFromCurrentTransformed();
        if (Motion1::isAxisHomed(Z_AXIS)) {
            setTmpPositionXYZ(currentPosition[X_AXIS],
                              currentPosition[Y_AXIS],
                              currentPosition[Z_AXIS] + oz - toolOffset[Z_AXIS]);
            moveByOfficial(tmpPosition, Motion1::moveFeedrate[Z_AXIS], false);
            waitForEndOfMoves();
        } else {
            currentPosition[Z_AXIS] += oz - toolOffset[Z_AXIS];
            updatePositionsFromCurrent();
        }
        toolOffset[Z_AXIS] = oz;
        updatePositionsFromCurrentTransformed();
    }
```

    The log output was:
        11:34:25.094 : N27 G28*38
        11:34:25.753 : Info:Autoleveling disabled
        11:34:25.753 : X:0.00 Y:0.00 Z:9.000 E:0.0000
        11:34:34.913 : setToolOffset: TOffXalt=7.00 TOffYalt=3.00 OxNeu=15.00 OyNeu=23.00   XYHomed!   CurrPosX=0.00 CurrPosY=0.00 TempPosX=8.00 TempPosY=20.00      TOffXNeu=15.00 TOffYNeu=23.00
        11:34:40.300 : setToolOffset: TOffXalt=15.00 TOffYalt=23.00 OxNeu=7.00 OyNeu=3.00   XYHomed!   CurrPosX=0.00 CurrPosY=0.00 TempPosX=-8.00 TempPosY=-20.00     TOffXNeu=7.00 TOffYNeu=3.00
        11:34:40.959 : Info:Autoleveling disabled
        11:34:40.959 : X:0.00 Y:0.00 Z:4.000 E:0.0000
        11:34:41.427 : SelectTool:0
        11:34:41.786 : setToolOffset: TOffXalt=7.00 TOffYalt=3.00 OxNeu=7.00 OyNeu=3.00   XYHomed!   CurrPosX=0.00 CurrPosY=0.00 TempPosX=0.00 TempPosY=0.00     TOffXNeu=7.00 TOffYNeu=3.00
        11:34:42.144 : X:0.00 Y:0.00 Z:4.000 E:0.0000

Interpretation of Log:
    TOffXNeu=15.00 TOffYNeu=23.00 is correct, but this position is not reached
    TOffXNeu=7.00 TOffYNeu=3.00 is correct, but this position is not reached
    
I think it is because in ...
    void PrinterType::officialToTransformed(float official[NUM_AXES], float trans[NUM_AXES])
    ... the toolOffset is added again which is previously subtracted.
    What I cannot explain, why does it depend on tool1 offset and
    why it works for zProbeOffset und Tool2Offset, if tool1Offset is 0.

repetier, 2020-10-09:
The first
```
            setTmpPositionXYZ(currentPosition[X_AXIS] + ox - toolOffset[X_AXIS],
                              currentPosition[Y_AXIS] + oy - toolOffset[Y_AXIS],
                              currentPosition[Z_AXIS]);
```
is correct. Here the - toolOffset[X_AXIS] compensates the + toolOffset[X_AXIS] from officialToTransformed so only ox/oy as difference stay which is the distance we need to move.
Afterwards we store them as official offsets since we should now be at that position:
```
        toolOffset[X_AXIS] = ox;
        toolOffset[Y_AXIS] = oy;
```
Then at the end updatePositionsFromCurrentTransformed(); copies the transformed position we are to official positions. If we would not have moved what we said we should now see the real position we are at.

How sure are you that positions are wrong? You wrote:
    zProbe pos was 8/20 NotOK should be 15/23
        probable calc X: -ProbeOffX + Tool1OffX (- (-15) + (-7) = 8)
        probable calc Y: -ProbeOffX + Tool1OffX (- (-23) + (-3) = 20)

But 8/20 is the correct position for the first move. It is correct because at that time the offset 7/3 are still present so to move to the correct official position must be 8/20. Later we fix the offsets to 15/23 and then officialToTransformed calculates we are still at 0/0. So from numeric logic the calculations seem correct.

So did you measure physically that positions are wrong? Or is that just from interpretation of numbers?

JoeScotty, 2020-10-09:
All values in the gray field "Test steps were..." are values taken with a ruler, mounted on X and Y axis, so physical values.

As far as I understand:
If Tool1 is activated by T0, the physical position has to move to 7/3,
If Tool2 is activated by T1, the physical position has to move to 10/4,
I ZProbe is activated,  the physical position has to move to 15/23.
Only those positions should happen in my test case.
The offical position ist 0/0 in every case.

You said "But 8/20 is the correct position for the first move". 
 
I think, 8/20 is only the distance between Tool1offset and ZProbeOffset,
but the position 8/20 should never happen.

But physically I measured 8/20 as a position.

Is it at the end only a question, if a command moves a certain distance (relative)
or a command move to a certain position (absolute)?

And thank you very much for your quick and detailed answer!

repetier, 2020-10-09:
Ok, I think you missed the most essential part and everything is correct in reality.
Tool 1 offset -7/-3
Tool 2 offset -10/-4
Z-probe offset -15/-23

what does firmware do? For all tools it computes the min and max offset. So for x that is -10.. -7 and for y it is -4 .. -3. The z probe is not considered here to not reduce printable area by bad position. Only the testable area gets reduced by z probe offset.

Now you home xy and position is selected so that you can switch between extruders without hitting the end stop. Let's only take x axis to make it more simple since also errors on y act identically.
Tool 1 and to have a difference of 3mm in offsets so depending on which is active after homing it moves 3mm or 0mm from the end stop. And here is your error I think. You assume it moves 7 or 10mm right? No it does not and the difference is error in position you think you have. You are irritated because if tool 1 had 0/0 the min value is defined by extruder 2 are also the moves. But that is just because difference with 0 is still the value.
So back to "bad" values. Homing with tool 1 active means offset is set to xoff = 7 and since it is the right extruder it does not move when hitting end stop (except set extra distance x move after home). So at position 0 you have offset 7 active. Activating z probe it moves 8mm to the right so offset is now 15 but position you measure is 8 since we did not move after x home.
Hope you see where you did go wrong in your logic? I know it is a bit hard to understand and also gave me a headache getting the math correctly.

JoeScotty, 2020-10-11:
Yes, I think I got it.
You are one step further in automatically maximizing the printable area than my thinking had been.
Thanks a lot for your fast and detailed answer!

Let me try it in my words:
The important groups of parameters when using offsets are:
1) all the distances between tool1 and toolx in X,Y,Z (and zProbe in X,Y)
2) the distances between end stop switching point and the position, where the most right tool (tool1??) is at the leftmost end of the printable area

The distances for 2) are the ENDSTOP_.._BACK_ON_HOME parameters
The distances for 1) are defined by absolute offset values, but just the relative distances are calculated in the firmware

The logical consequence is, that

        Tool1 Offset -4/-20
        Tool2 Offset -30/-21
        ZProbe Offset -20/+5

        is handled in the same way as for example
        Tool1 Offset -24/-20
        Tool2 Offset -50/-21
        ZProbe Offset -40/+5

        and is also handled in the same way as
        Tool1 Offset 0/0
        Tool2 Offset -26/-1
        ZProbe Offset -16/+25

which might be irritating if one is not aware of the internal calculations.
I remember the offset drawing in configurator of V1.x where all offsets are reference to 0/0 and tools and zprobe around it.

So on one hand it would be more intuitive just entering relative distances between tools and ZProbe positions,
on the other hand it is useful to have parameters which are backwards compatible.

So my proposal is to document this as a rule set like:
    All Offset values are handled as relative distances to tool1
    It is recommended to define Tool1 offset allways as 0/0
    All other tool and ZProbe Offsets are the distance between the Toolx / ZProbe and Tool1
    To define, where Tool1 should be positioned after homing, take the parameter ENDSTOP_.._BACK_ON_HOME,
    which defines the distances in X, Y, Z between end stop switching point and the official position 0/0/0.
    And a drawing ;-)

Regards!

repetier, 2020-10-12:
First z probe offset is not taken into computation for extreme offsets als already mentioned. Instead for z probing the allowed area gets reduced if needed.

Also for some systems like delta the 0,0 position is defined by a mathematical requirement. Only for simple cartesian motion it is freely selectable.

Apart from this I think you have understood it correctly.

closed 2020-10-12 by repetier