repetier/RepetierFirmware #992: Dev2: add M218 function

GenaUser, 2020-10-12:
Would it be possible to add the Gcode M218 function to allow heads offsets changes directly from a touch screen or from any other interface?

Thank's in advance for analyzing this possibility

repetier, 2020-10-12:
Oh you were using this firmware with server:-) Go on touch screen to settings->eeprom and you can already change it.

GenaUser, 2020-10-12:
Yes, I agree with you ;-)
The server solution is useful for many parameters. But when you change the nozzle and need to adjust the offsets, it is not always practical to go to a computer to change the values.
For example, on my printer's touch screen, I cannot use the console for this task, neither the offset menu because they work with the M218 command.
To work around this problem, I have to use a second LCD screen with the Repetier interface to make the corrections

[image]

[image]

[image]

repetier, 2020-11-06:
I have added M218 marlin style so hope it works now. Only difference is that we do not add echo: before the output.

closed 2020-11-06 by repetier