repetier/RepetierFirmware #1071: Beeper doesn't stop beeping in dev2

nukem, 2021-03-27:
I am having a hard time configuring the beeper using a standard IO pin.

I followed the documentation
```
// Beeper
#if BEEPER_PIN > -1
IO_OUTPUT(IOBeeperMain, BEEPER_PIN)
BEEPER_SOURCE_IO(MainBeeper, IOBeeperMain)
#endif
```

The beeping works, it just it doesn't stop.

I have to add this to my Printer.cpp to make it work

```
void Printer::defaultLoopActions() {
    beepers[0]->process();
```

I only have one beeper so that fix is ok for me.

Is there something I need to do in config to make sure that fast8_t BeeperSourceBase::process() is called? 
I don't see it being referenced anywhere in the code to make sure the beep loop ends.

nukem, 2021-03-27:
It seems if I don't have a servo on my configuration the beeper->process() is never included on the interrupts.

repetier, 2021-03-27:
Ok thanks to that hint it was a easy one. If you look at io_beeper.h

```
#elif IO_TARGET == IO_TARGET_SERVO_INTERRUPT // IO_TARGET_PERIODICAL_ACTIONS

#define BEEPER_SOURCE_IO(name, IOPin) \
    name.process();

#define BEEPER_SOURCE_PWM(name, PWMPin) \
    name.process();

```

you see that process gets called in servo timer. After checking the code I saw that when you have no servo the servo timer is also not available. So I changed servo timer to enable when servo or beeper counter is present. Should now also work with official version. Thanks for the info.

closed 2021-03-27 by repetier

nukem, 2021-03-27:
It will have compile errors if no servo is defined.

repetier, 2021-03-27:
After removing my servos I got no compile error. What errors do you get?

nukem, 2021-03-27:
in Configuration.h

`#define NUM_SERVOS 0  `

```

In file included from src\io/redefine.h:58:0,
                 from src\Configuration.h:194,
                 from src\Repetier.h:302,
                 from src\Configuration.cpp:29:
src/Configuration_io.h: In member function 'void Servo1Class::init()':
src\io/io_servos.h:38:18: error: 'servoMicroseconds' is not a member of 'HAL'
             HAL::servoMicroseconds(slot, neutral, 1000); \
                  ^
src/Configuration_io.h:232:1: note: in expansion of macro 'SERVO_ANALOG'
 SERVO_ANALOG(Servo1, 0, Servo1Pin, 500, 2500, 1050)
 ^~~~~~~~~~~~
src/Configuration_io.h: In member function 'virtual void Servo1Class::setPosition(int, int32_t)':
src\io/io_servos.h:49:18: error: 'servoMicroseconds' is not a member of 'HAL'
             HAL::servoMicroseconds(slot, pos, timeout); \
                  ^
src/Configuration_io.h:232:1: note: in expansion of macro 'SERVO_ANALOG'
 SERVO_ANALOG(Servo1, 0, Servo1Pin, 500, 2500, 1050)
 ^~~~~~~~~~~~
In file included from src/io/redefine.h:58:0,
                 from src/Configuration.h:194,
                 from src/Repetier.h:302,
                 from src\PrinterTypes\Printer.cpp:19:
src/Configuration_io.h: In member function 'void Servo1Class::init()':
src/io/io_servos.h:38:18: error: 'servoMicroseconds' is not a member of 'HAL'
             HAL::servoMicroseconds(slot, neutral, 1000); \
                  ^
src/Configuration_io.h:232:1: note: in expansion of macro 'SERVO_ANALOG'
 SERVO_ANALOG(Servo1, 0, Servo1Pin, 500, 2500, 1050)
 ^~~~~~~~~~~~
src/Configuration_io.h: In member function 'virtual void Servo1Class::setPosition(int, int32_t)':
src/io/io_servos.h:49:18: error: 'servoMicroseconds' is not a member of 'HAL'
             HAL::servoMicroseconds(slot, pos, timeout); \
```

nukem, 2021-03-27:
ok I'll remove the servo in configuration_io.h but it will still give errors because I don't have an initial servo config before.

nukem, 2021-03-27:
After removing all servo thing on my config.

```
src\boards\due\HAL.cpp:999:33: error: 'unsigned int HAL::servoTimings [4]' is not a static data member of 'class HAL'
 unsigned int HAL::servoTimings[4] = { 0, 0, 0, 0 };
                                 ^
src\boards\due\HAL.cpp:1003:74: error: no 'void HAL::servoMicroseconds(uint8_t, int, uint16_t)' member function declared in class 'HAL'
 void HAL::servoMicroseconds(uint8_t servo, int microsec, uint16_t autoOff) {
                                                                          ^
src\boards\due\HAL.cpp: In function 'void TC6_Handler()':
src\boards\due\HAL.cpp:1029:26: error: 'servoTimings' is not a member of 'HAL'
                     HAL::servoTimings[servoId] = 0;
                          ^~~~~~~~~~~~
src\boards\due\HAL.cpp:1034:22: error: 'servoTimings' is not a member of 'HAL'
             if (HAL::servoTimings[servoId]) {
                      ^~~~~~~~~~~~
src\boards\due\HAL.cpp:1036:33: error: 'servoTimings' is not a member of 'HAL'
                 interval = HAL::servoTimings[servoId];
                                 ^~~~~~~~~~~~
Compiling .pio\build\digix\src\communication\GCodes.cpp.o
*** [.pio\build\digix\src\boards\due\HAL.cpp.o] Error 1
```

repetier, 2021-03-27:
I see I have no beepers in my config :-) Will test with beeper.

repetier, 2021-03-27:
You need to put
#define NUM_BEEPERS 1
in configuration.h before the includes to get it work for now. Need to find a better solution, but might not be possible. Same problem why NUM_SERVOS is at the top.

repetier, 2021-03-27:
Ok found a solution. Just needed to remove the #if condition in .h file. As long as no one uses it when disabled there will be no missing implementation as it seems.

nukem, 2021-03-27:
I have it working and removed my fix by just defining a servo.

Others will get the same compile errors if they don't have a servo and have a beeper.

Maybe define fake servos by default on the default config and put notes on the configuration files.

repetier, 2021-03-27:
Not needed any more with latest commit.

nukem, 2021-03-27:
It does compile, but if you don't have a servo the Timer don't get set properly and the interrupts don't happen and buzzer don't stop.