repetier/RepetierFirmware #1079: dev2: Homing causes motion on E axis (extruder)

kyrreaa, 2021-05-23:
Full reset on board, homing (G28) and issuing "G92 E2" motion will happen on extruder when issuing new homing.
The motion does not reset offset and motion now happens on every homing. This makes recovering prints very hard as E-axis must be cleared before homing to recover position, and restored to allow print to continue...
(Running Core-XYZ setup with 4 motors.)

repetier, 2021-05-24:
Sorry, but I do not really understand what you are trying to say. You mean after G92 E2 a new homing will move extruder these 2mm?

kyrreaa, 2021-05-24:
Yes, or retracts, at high speed.

repetier, 2021-06-06:
Sorry for the long delay, was quite busy.

I have checked it. I could not reproduce the retract problem, but a saw that E offset does not get reset. I fixed that now hoping it helps. 

E gets reset if G28 has no parameter or contains E0. It is set with priority 0 so if the problem happens already on X homing you need to set x home priority to 1 at least.

Hope it works now, otherwise please tell me very exact which of the many homing moves does the retract. I have no core xy so if there is something special that might explain it, also it just runs the same simple home commands and xyz mapping does not include e so I expect no difference.