repetier/RepetierFirmware #1083: Typo maybe.....

johnnysmithy852456, 2022-01-31:
https://github.com/repetier/Repetier-Firmware/blob/2bbda51eb6407faf29a09987fd635c86818d32db/src/ArduinoDUE/Repetier/Configuration.h#L1819

Hi Correct me if I'm wrong, Is this a this typo FAN2_PIN ORIG_FAN2_PIN should be FAN_PIN ORIG_FAN2_PIN .
I was trying to get the second part cooling fan running on Luc's [Repetier-Firmware-4-Davinci](https://github.com/luc-github/Repetier-Firmware-4-Davinci)
https://github.com/luc-github/Repetier-Firmware-4-Davinci/blob/c6cd7a4b72a4a05405694e6eebb3e015aae464ec/src/ArduinoDUE/Repetier/Configuration.h#L1921

Because the davinci 1.0a was only shipped with one extruder heater cooling fan there is a option to add a part cooling fan on the main board with a dedicated GPIO pin ready to go.

I couldn't get it to work until I change it to FAN_PIN ORIG_FAN2_PIN.
Thanks

repetier, 2022-01-31:
FAN2_PIN is used for a second fan in case you have 2 fans that should get controlled. So that is correct. You then use FAN_PIN plus FAN2_PIN for M106 P1 Sxxx.

You can also use one of them for extruder cooling. These are actually just variables and it all depends on how you set up your configuration. When you set 2 cooling fans they are used like that. If only one, you can use the second to assign it for extruder cooling. Guess you had FAN_PIN assigned for cooling extruder so you needed to override it that way. But that is just the special case for wiring and config update.

closed 2022-01-31 by repetier

johnnysmithy852456, 2022-02-04:
Hi Thank you for reply
Yes I have one fan that turns on with the extruder heater. (On constantly) and the second thats controlled by the M106 for part cooling.
I understand thank you for taking the time to explain.