repetier/RepetierFirmware #1090: Crush while connecting my FlsunQ5 printer in a Surface PRO 7

SquintPlug7, 2023-01-25:
It crush every itme i m pressing the button "connect" idk why... i dont think it is an app too heavy to handle for my pc ... it s a surface pro 7 and to be honest it can handle much heavier app

repetier, 2023-01-25:
Firmware runs on printers not on surface and I don't think you have installed it since it has no connect button.

So what software are you talking about?
If you mean  Repetier-Server make sure you have not version 1.4.4 which has an issue with non utf-8 answers that can cause crash on connect.

closed 2023-01-25 by repetier

SquintPlug7, 2023-01-25:
Oh thankyou, i was meaning repetier Host by the way not the server.... but to be honest i think i have the 1.4.4  server version. So what can i do to change it?

repetier, 2023-01-25:
Open the web interface of  Repetier-Server ( http://localhost:3344 ) and go to check updates. There is a autoupdater that will upgrade when clicked.

Since we install server by default that might in deed be the case.