repetier/RepetierFirmware #1091: Issue Compiling for Melzi V2 (Wanhao Duplicator i3) Repetier V1.0.4

MedievalApple, 2023-02-20:
Whenever I try to compile `Firmware version 1.0.4` for the Melzi v2 I'm always met with this error `/Repetier.ino.elf section .text' will not fit in region text', and ld.exe: region text' overflowed by 11382 bytes`. I've tried to get the size down by disabling features in the config and that got the overflow down a bit but never enough to compile.

The processer I have it set to compile to in Arduino is the `ATMEGA 1284P (16 MHz)`

The Arduino Board Lib I'm using is `https://github.com/Lauszus/Sanguino`

The Arduino IDE version is `2.0.3`

The Config file I'm trying to compile with `https://gist.github.com/MedievalApple/7716978c2d70796c931173ae876a5b45` (Made with the Repetier Firmware configuration tool )

repetier, 2023-02-20:
Text area is limited to 64kb as it must be in first 64kb with jump tables and some things, so you need to reduce everything in it. Select at most one language for lcd. Also disabling eeprom saves a lot but might not be desired. sd card also uses quite some ram. arc support might be disabled. But in the end you must decide what you really can miss until you are below allowed usage.

MedievalApple, 2023-02-20:
Thank you disabling eeprom support did indeed solve the compile problem. 😄

closed 2023-02-20 by MedievalApple