repetier/RepetierFirmware #1092: x, y, z axis only move in one direction

abaker3003, 2023-03-20:
I fixed this man-made 3d printer, was unable to obtain the original config.h file for it but I managed to update the firmware from 0.92.3 to 1.0.4. Now im having the axis issue. 
My board is AZTEEG X3 PRO (Atmel ATmega 2560)
I have attached my repetier folder with my configuration.h and pins.h files
The professor that made this 3d printer is reluctant to help me and I'm only a CS (former EE) major. I knew nothing about 3d printers until now through all the trail and errors.

please help ): 

[Repetier-Firmware-2023-03-19 (7).zip](https://github.com/repetier/Repetier-Firmware/files/11012759/Repetier-Firmware-2023-03-19.7.zip)

repetier, 2023-03-20:
Check end stop signal with M119 and compare it to real state. If you can only move away from end stop it is normally just end syop being high preventing that direction. Invert end stop logic in that case.

repetier, 2023-03-20:
So your printer has really 6 endstops? Most have only end stops in homing direction. Anyhow you see all min endstops have a setting making them high preventing moves in that direction. Go into the config editor and select the inverted signal of what you now have for endstop and it should be low and only high when triggered.

https://www.repetier.com/firmware/dev/index.php

[image]

abaker3003, 2023-03-20:
> You don't often get email from ***@***.*** Learn why this is important<https://aka.ms/LearnAboutSenderIdentification>

I’m dead lol duly noted!! Like I said, I know nothing about 3d printers. So just by what you said, it’s safe for me to assume this thing doesn’t have 6 end stops. A lot of the configurations I saw before installing it I had no clue what to put so I skimmed through it. That could also be why I’ve been getting a temperature sensor error when trying to head up the bed now. I’ll run through the configurations again more thoroughly in the morning and post an update on the Issue. Thank you so so much

https://www.repetier.com/firmware/dev/index.php

[image]<https://user-images.githubusercontent.com/933919/226286127-449332f0-8493-430e-b77c-04f72b461c92.png>