repetier/RepetierFirmware #1095: Issue on Spiderking 407 motherboard, couldn't get any axis to move

Maddy5075, 2023-09-13:
Use case: 

 I use external stepper drivers (closed loop) I would like the motors configured like this.  
 Mot1 X ,  Mot2 X2 ,  Mot3 Y ,  Mot4 Y2,  Mot5 Y3 ,  Mot6 Z ,  Mot7 Z2 ,  Mot8 Z3,  Mot9 Z4 ,  Mot10 E0.  
 I might not use all the motors on the axis but it gives me the option.  
 The only two functions I really need are filament out on Ymin endstop and "door open" on Xmax endstop.  
 A door open trigger pauses the printer.

[spider_king_407.json.txt](https://github.com/repetier/Repetier-Firmware/files/12598798/spider_king_407.json.txt)
[PinNamesVar.h.txt](https://github.com/repetier/Repetier-Firmware/files/12598804/PinNamesVar.h.txt)
[variant.cpp.txt](https://github.com/repetier/Repetier-Firmware/files/12598805/variant.cpp.txt)
[variant.h.txt](https://github.com/repetier/Repetier-Firmware/files/12598806/variant.h.txt)
[ldscript.ld.txt](https://github.com/repetier/Repetier-Firmware/files/12598807/ldscript.ld.txt)
[PeripheralPins.c.txt](https://github.com/repetier/Repetier-Firmware/files/12598808/PeripheralPins.c.txt)

[platformio.ini.txt](https://github.com/repetier/Repetier-Firmware/files/12598816/platformio.ini.txt)

[Configuration.h.txt](https://github.com/repetier/Repetier-Firmware/files/12598831/Configuration.h.txt)
[Configuration_io.h.txt](https://github.com/repetier/Repetier-Firmware/files/12598832/Configuration_io.h.txt)

Able to generate the bin file and when flashed we see this behaviour

We then tried to connect to the printer but had problems because the door open was constantly being triggered. I couldn't resolve it so I've commented it out in the firmware for the time being. 

We can heat the hot end and heated bed but there is an issue with the sensors. If we heat the hotend only for some reason the heated bed temperature rises as well identical to the hotend even though the heated bed isn't actually heating up. We have the hotend thermistor connected to TE0 and heated bed to T5(Tbed). If we heat the bed only the hotend thermistor goes up as well identical to the bed temp even though the hotend is not heating. 

If we try to home the x axis(MOT1) the extruder motor turns(MOT10). If We connect the x axis to MOT10 and press home the x axis moves away from endstop. Trying to home Y or Z axis only produces "Fatal Error" and no motor movement. Sending an M119 gives the endstop state as in image. If I hold any endstop in and send a M119 nothing changes so the endstops don't seem to be working.

[image]

[image]

Maddy5075, 2023-09-14:
[spider_king.h.txt](https://github.com/repetier/Repetier-Firmware/files/12605070/spider_king.h.txt)

repetier, 2023-09-14:
I see you are tying to add a new motherboard, which is always a hard task even if the processor is supported.
First thing you need to do is always watch timer usages. We use several internally for timings and none of them can then be used for pwm signals to fan/heater. in STM32F4/hal.h you see how they are using fallbacks if not defined in motherboard definition.
```
#ifndef MOTION2_TIMER_NUM
#define MOTION2_TIMER_NUM 6
#endif
// Beware! On STM32F4's, TIM10's IRQHandler is shared with TIM1's!
// Both timers will interrupt with the same function!
#ifndef PWM_TIMER_NUM
#define PWM_TIMER_NUM 10
#endif
#ifndef MOTION3_TIMER_NUM
#define MOTION3_TIMER_NUM 7
#endif
// Timer 13 does crash firmware at least for rumba32!
#ifndef SERVO_TIMER_NUM
#define SERVO_TIMER_NUM 9
#endif
#ifndef TONE_TIMER_NUM
#define TONE_TIMER_NUM 11
#endif
```
In peripherals you see which timers are used for pwm and often you have multiple timer as option. So make sure no overlap exists to create unwanted interference.

Also check the analog inputs used to be complete. The rest is quite straight forward if you use correct pin numbers and they are not used for some hardware function like spi/i2c/uart. I can not check that since I do not have the board or time to that. But if you have special questions, let me know.

One trick for endstop testing, send
M111 S71
and you see every signal change on screen, no M119 needed to query all the time.

For motor testing send
G1 P0 or P1 to disable boundary check and configure to move without homing. Then you can try movements to be in correct direction first. Especially if you have checking endstops disabled for regular moves. So that way you only test motor pins and later you can add endstops for homing, but signaling must be correct.