repetier/RepetierFirmware #1097: Issue with missed steps, incorrect positioning after collision

fox88x, 2024-03-14:
Hello everyone,

I have upgraded my flsun SR printer to Klipper with a Raspberry Pi. Overall, the result is good but I have a major issue with homing after a collision. If for some reason during a print a collision occurs and the steppers lose steps, in subsequent prints there remains a sort of memory of the lost steps with potentially disastrous consequences (scratched bed).
(The drivers are the original ones: TMC2209)

I conducted the following test:

1)Printer calibration, initially cold, while bed leveling and adjusting the z_offset setting the working temperature

     - Z_OFFSET_CALIBRATION
     - DELTA_CALIBRATION
     - ENDSTOPS_CALIBRATION (ENDSTOP_PHASE_CALIBRATE)
     - BED_LEVELING
     - Z_OFFSET ADJUST

2) Verification of successful calibration with a test print: first layer test
3) After homing, I perform a movement and manually block the printer's effector causing the steppers to lose steps
4) I perform a homing and repeat the test print

At this point, the positioning is incorrect, the print scratches on the bed and/or is lifted.
I believe I have only encountered this problem with Klipper, with Marlin (original) I don't recall this issue.
I think it's related to ENDSTOP_PHASE_CALIBRATE but it's just an idea and after much research I wouldn't know how to intervene anyway. Can you help me?

Thank you

repetier, 2024-03-14:
You are in the wrong project, this is for Repetier-Firmware not klipper.

closed 2024-03-14 by repetier

fox88x, 2024-03-14:
sorry, thanks for the correction